balena-io/etcher #3068: Etcher not flashing on mac

happyduck1, 2020-02-14:
- **Version 1.5.76** 
- **Operating system and architecture:Mac os** 
- **rasberry pi image file** 
Hi, when installing the rasberry pi .img file (came from .zip) 
I select the .img file, then the drive I want then click flash, enter my mac password, then it goes back to the last image.
[image]
[image]
[image]
[image]

uYtLD9zS, 2020-02-14:
I am having the same problem.
Operating system and architecture:MacOS Catalina
pfSense image file

uYtLD9zS, 2020-02-14:
@happyduck1 - make sure you are logged in to an administrator account

KondorB, 2020-02-20:
Have the same problem here. Earlier versions were asking for admin login and password and worked fine afterwards. So, why this version just assumes running from admin?

lurch, 2020-02-20:
More info in #3067

rubysims, 2020-03-12:
what is an a administrator account

ieaves, 2020-05-25:
@lurch #3067 seems to indicate an issue on MacOS, am I reading that correctly? I'm having a similar issue to all of those above still and have exhausted the debugging steps that appear available. Just trying to make sure I haven't missed anything before giving up.

lurch, 2020-05-26:
@ieaves I'm not an Etcher developer (and I don't have a Mac), but AFAIK the most up-to-date info about these MacOS issues is in #3107 

@thundron Maybe this should be closed as a duplicate of #3107 ?

thundron, 2020-05-29:
Definitely, refer to the other issue for more information

closed 2020-05-29 by thundron