balena-io/etcher #3078: 403 Forbidden error on package repository

hollec, 2020-02-23:
- **Etcher version:** 
1.5.76
- **Operating system and architecture:**
Ubuntu 19.10 (x64)
- **Image flashed:** 
N/A
- **Do you see any meaningful error information in the DevTools?** 
N/A

When performing an apt-get update, I receive a 403 Forbidden error from the Etcher Debian package repository.

> ...
> Err:18 https://dl.bintray.com/etcher/debian stable Release               
>   403  Forbidden [IP: 54.201.242.72 443]
> ...
> E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
> N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> N: See apt-secure(8) manpage for repository creation and user configuration details.

There were previously no issues with the repository and the repository line and GPG key matches the [README](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64).

`deb https://deb.etcher.io stable etcher`

legolas-salogel, 2020-02-23:
hi, 
Debian and Ubuntu based Package Repository (GNU/Linux x86/x64) is down

gratefulfrog, 2020-02-23:
Same for me ... XUbuntu 18.04

Dude4Linux, 2020-02-23:
Same for me ...  Files are still listed at https://dl.bintray.com/etcher/debian/ but when I try to retrieve anything I get
```
Forbidden!
```
Either a permissions problem or a payment issue.

cowboysmall, 2020-02-24:
Same for me - with rpm-based repository.

CanonCan, 2020-02-25:
Same for me with Fedora workstation 30

mboudreau, 2020-02-25:
Same here, getting a 403 when apt-get tries to get this file: https://dl.bintray.com/etcher/debian/dists/stable/Release

ddt3, 2020-02-25:
Same here, but as this is not local that is not a surprise...

tomreyn, 2020-02-25:
I think we have sufficiently established that this affects everyone trying to download files from URLs starting `https://dl.bintray.com/etcher/`.

**If you are also affected by this, please just subscribe to this bug. There is no need to post another "me too" comment.** Thanks.

DinStamou, 2020-02-25:
Same here, Debian Testing:
`E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.`
I believe it has to do with the Release file.

thundron, 2020-02-25:
We're working on fixing the issue, I'll update this issue as soon as it's solved

ToomaiO, 2020-02-27:
The central problem is pointed from 
https://github.com/balena-io/etcher/issues/3078#issuecomment-590128779
it looks as a rigth toi access problem.

lurch, 2020-02-27:
BTW @thundron I just noticed that https://bintray.com/etcher/debian says "Resin.io Debian Packages" :eyes: 
Maybe it should be changed to "Etcher DEB Packages" to make it more in line with https://bintray.com/etcher/redhat ?

CanonCan, 2020-02-27:
Is there any updates? it’s been more than two days since it’s been down

thundron, 2020-02-28:
We're in touch with Bintray to get this solved, unfortunately we can't provide any ETA

giupardeb, 2020-02-28:
in fedora 31 I have been the same problem for a few days

Ninoh-FOX, 2020-02-29:
same error in debian 9/10:

Err:11 https://dl.bintray.com/etcher/debian stable Release   
  403  Forbidden

ijash, 2020-03-02:
same problem

enguerran-popup, 2020-03-02:
@thundron Could you lock this issue for contributors only? So we can avoid the "same here" spam.

thundron, 2020-03-02:
@siucdude not yet, again we're in constant touch with Bintray (the download provider) to get this solved ASAP.
In the meantime, please update/install through the release page
Locking down for now, we'll update the issue as soon as this is solved. It's a matter of communication unfortunately so we can't give have an ETA on that

thundron, 2020-05-04:
Solved, following on https://github.com/balena-io/etcher/issues/3133 for another issue

closed 2020-05-04 by thundron