balena-io/etcher #3080: Can

NelDav, 2020-02-24:
- **Etcher version:** 1.5.79
- **Operating system and architecture:** Windows 10
- **Image flashed:** 2020-02-13-raspbian-buster-lite
- **Do you see any meaningful error information in the DevTools?** 
> GET file:///C:/Users/DN/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png net::ERR_FILE_NOT_FOUND

I think etcher should not crash because of missing image

lurch, 2020-02-24:
Possibly a duplicate of #3039 ? I wonder why more people aren't getting this error? :man_shrugging:

thundron, 2020-02-25:
When does this happen?

NelDav, 2020-03-07:
It happens, directly bevore flashing.

I solved the problem by adding Balena as exception in "Allow an app through Controlled folder acces". (Settings->Update & Security->Windows Security->Virus & threat protection->Manage Settings->Controlled folder access->Allow an app through Controlled folder acces)

lurch, 2020-03-08:
Ahhhh. Seems like tweaking that setting somehow fixes many Etcher problems? https://github.com/balena-io/etcher/search?q=controlled+folder+access&type=Issues

wahyudwikrisnanto, 2020-04-30:
Im still had the problem even my setting on controlled folder access is turned off

thundron, 2020-05-06:
@wahyudwikrisnanto have you tried opening as admin?

zvin, 2020-05-06:
fixed in 1.5.84

closed 2020-05-06 by zvin