balena-io/etcher #3082: Cross compiling instructions for arm64 (v8)

ickam, 2020-02-24:
Hi! This is a bit unusual, I admit, but I really need etcher to be installable on an arm v8 board. The board has two storage options and I need to flash one of those storages. Because I'm physically disabled, taking the storage out to flash it with a prebuilt etcher is not really an option.
So I am kindly asking for instructions to cross-compile etcher for armv8. I have a kubuntu 19.10 available, I downloaded the newest version of linaro and added it to my PATH in .bashrc, but since I've never cross-compiled, I'd really appreciate a step by step instruction to build etcher for my Odroid c2.
After I've done it, I am more than happy to share the binary built.
Thanks in advance.

lurch, 2020-02-24:
You probably want to keep an eye on #2870 ? :eyes:

zvin, 2020-03-07:
Hello,
On a debian or ubuntu box:
```
sudo apt-get install python build-essential  # maybe some other stuff is needed
git clone --recursive https://github.com/balena-io/etcher.git
cd etcher
make electron-develop
PATH=$(pwd)/node_modules/.bin:$PATH electron-builder --linux deb --arm64
```
That should build and `aarch64` `deb` package of Etcher in `dist/balena-etcher_1.5.79_arm64.deb`.
Please refer to https://www.electron.build/configuration/linux#LinuxConfiguration-target for other options.

rradar, 2020-03-21:
Maybe it's easier for you @ickam (and does save a lot's of time + bandwidth) to use a ready made binary. I'm not 100% sure but it could be that a [armhf](https://gitlab.com/bztsrc/usbimager/-/blob/binaries/usbimager_1.0.0-armhf.deb) or a [armv7l](https://gitlab.com/bztsrc/usbimager/-/blob/binaries/usbimager_1.0.0-armv7l-linux-x11.zip) version works for you. It's just around 300Kb in size (~ factor 1000 smaller than etcher) but performance much better for me (which comes specially handy on arm devices).

closed 2020-03-21 by zvin