balena-io/etcher #3083: repo fails since latest update 1.5.76 fedora 30

CanonCan, 2020-02-24:
- **Etcher version:** repo
- **Operating system and architecture:** fedora 30 x86_64 kernel 5.4.19

running dnf update fails with this error 

sudo dnf update
etcher-rpm                                                                    
Errors during downloading metadata for repository 'etcher-rpm':
  - Status code: 403 for https://dl.bintray.com/etcher/redhat/repodata/repomd.xml (IP: 52.11.162.82)
Error: Failed to download metadata for repo 'etcher-rpm': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried

PeterSprague, 2020-02-24:
Same, on Fedora 31

JakeTrock, 2020-02-24:
I am also experiencing this same problem with your Debian ppa. Would it be possible to release this to the snap store instead?
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package balena-etcher-electron is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'balena-etcher-electron' has no installation candidate
`

GammaGames, 2020-02-24:
Looks like a duplicate of #3078

lurch, 2020-02-25:
> Would it be possible to release this to the snap store instead?

For snap, see #415

CanonCan, 2020-02-25:
> Looks like a duplicate of #3078

Sorry I missed that was skimming the issues when couldn’t find it I posted here but either way no solution in sight it seems

thundron, 2020-02-25:
Thank you for the report, working on it, please refer to the other issue

closed 2020-02-25 by thundron