balena-io/etcher #3084: update etcher using 'apt update' gives an error on Linux Mint The repository 'https://deb.etcher.io stable Release' does not have a Release file.

obeewaan, 2020-02-25:
- **Etcher version:*1.5.76* 
- **Operating system and architecture:*amd64 x86_64* 
- $ sudo apt update
Ign:15 https://deb.etcher.io stable InRelease     
Err:17 https://deb.etcher.io stable Release
  403  Forbidden
Reading package lists... Done
E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

rafabiela, 2020-02-25:
same problem

paolantoni, 2020-02-25:
either with debian 10.

jjogeva, 2020-02-25:
I can confirm the same issue on Ubuntu 18.04.

I am also fairly sure of the **actual problem** that has nothing to do with the OS.
It seams that someone has changed file permissions on https://dl.bintray.com/etcher/debian/dists/stable/ such that the files Release and Release.gpg are not readable by the web server. So the files are present but since the server sends 403 then the package manager just tells you that they are missing.

**Solution:** Could the person managing the server make these two files readable to the web server?

lollofra, 2020-02-25:
Same issue with Fedora (rpm based distro):

$ sudo dnf update etcher
etcher-rpm                                       21  B/s |  10  B     00:00    
Errors during downloading metadata for repository 'etcher-rpm':
  - Status code: 403 for https://dl.bintray.com/etcher/redhat/repodata/repomd.xml (IP: 52.28.187.224)
Error: Failed to download metadata for repo 'etcher-rpm': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
Ignoring repositories: etcher-rpm

NathanHannon, 2020-02-25:
Same on Pop!_OS 19.10

etchichury, 2020-02-25:
There is already an issue for this problem: #3078. 
They say they are working on it - https://github.com/balena-io/etcher/issues/3078#issuecomment-590819461
You can subscribe to that issue and wait for an update.

GammaGames, 2020-02-26:
Would be cool if they pinned the original issue to prevent dupes

thundron, 2020-02-26:
As mentioned by @etchichury , please follow the other issue, thanks for the patience

closed 2020-02-26 by thundron

SebastianCaballero, 2020-12-25:
The solution if you want to remove Etcher (the package) in order to generate the update of the packages is:
`sudo nano /etc/apt/sources.list`
remove those lines that contains the hhtp:// domain of the problem. Then CTRL+X, yes and ENTER.
Now you can `sudo apt update`.

krishnaTORQUE, 2021-05-03:
I am facing the same problem 

Error
```
Err:3 https://dl.bintray.com/etcher/debian stable InRelease                                                                                                                        
  403  Forbidden [IP: 34.214.135.19 443]
Reading package lists... Done                                                                                                                                                      
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 34.214.135.19 443]
E: The repository 'https://deb.etcher.io stable InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

lurch, 2021-05-03:
See #3501

rileymintable, 2021-05-03:
While @SebastianCaballero mentioned a possible solution to the issue, I discovered mine by commenting out Balena.io in /etc/apt/sources.list.d and finding the repo .list directly there. 

sudo vim /etc/apt/sources.list.d/balena-etcher.list

IamPhytan, 2021-05-03:
Thanks @rileymintable ! Your answer helped !

hugocortes, 2021-05-07:
in case anyone wants to add the apt source without using their provided script as is my preference, here are my steps:

1. add their gpg key
```sh
wget -qO - https://dl.cloudsmith.io/public/balena/etcher/gpg.70528471AFF9A051.key | sudo apt-key add -
```
2. modify existing repo, mine was under `/etc/apt/sources.list.d/etcher.list` and replace with this line (may vary depending on your system)
```
deb https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu xenial main
```
I got this deb line from this url that returns the deb package and is used by their script:
```
https://dl.cloudsmith.io/public/balena/etcher/config.deb.txt?distro=${distro}&codename=${codename}&version=${version}&arch=${arch}

# mine:
https://dl.cloudsmith.io/public/balena/etcher/config.deb.txt?distro=ubuntu&codename=xenial&version=20.10&arch=amd64
```
3. optional - you can remove the previous gpg key as well
```shh
sudo apt-key del 379CE192D401AB61
```

vasigorc, 2021-06-25:
For Fedora (`dnf` package manager) to disable `etcher-rpm` repository use:
```shell
dnf config-manager --set-disabled etcher-rpm
```