balena-io/etcher #3085: Debian package repo: missing Release file

KristianHolsheimer, 2020-02-26:
I just did a `sudo apt update` and it gave me this warning message:
```bash
$ sudo apt update
```
> ...
> E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
> N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> N: See apt-secure(8) manpage for repository creation and user configuration details.

Would you be able to have a look at this?

Thanks!

jdmg94, 2020-02-26:
I'm also having this issue on ubuntu

GammaGames, 2020-02-26:
Duplicate of #3078

monkeypants, 2020-02-27:
I'm not sure it's missing, I think the permission is wrong. See

https://dl.bintray.com/etcher/debian/dists/stable/Release

returns 403 not 404.

rbnmndz, 2020-02-27:
Yes, denied access (forbidden)

Err:38 https://dl.bintray.com/etcher/debian stable Release                                                                                                                
  403  Forbidden [IP: 52.29.129.139 443]

And also the releases are out of date, the last is balena-etcher-electron_1.5.76

jrodd96, 2020-02-27:
Also having this issue on Pop! OS 19.10 when running `sudo apt-get update`:
`Err:19 https://dl.bintray.com/etcher/debian stable Release                   
  403  Forbidden [IP: 52.11.162.82 443]`

thundron, 2020-02-27:
Working on it, please refer to the other issue #3078 . Closing

closed 2020-02-27 by thundron

mateMathieu, 2020-03-03:
Thank you thundron!