balena-io/etcher #3088: 1.5.79 on Windows 10, 1909, 64-bit fails to start

b-morgan, 2020-02-27:
-1.5.79
- Windows 10, 1909, 64-bit
- N/A
- N/A

The install completes but clicking on the icon opens multiple processes and no program window is displayed. The portable version behaves the same way. Using Run as administrator with the portable version is the same.

I believe 1.5.56 was the last version that worked for me on Windows 10 1903. Now, 1.5.56 portable creates a blank program window.

lurch, 2020-02-27:
Sounds similar to #1866 ?

b-morgan, 2020-02-27:
I agree, it does sound similar. I reviewed that issue and didn't see any real solution except to note that this problem has been around since 2017.

thundron, 2020-02-28:
Hi, what graphic card do you have in your pc? Is it integrated only or do you also have a dedicated card?

b-morgan, 2020-02-28:
[NVIDIA System Information 02-28-2020 07-13-05.txt](https://github.com/balena-io/etcher/files/4267405/NVIDIA.System.Information.02-28-2020.07-13-05.txt)

Ctrl-Shift-I output (or lack thereof):

Dialog box:

DevTools was disconnected from the page.

Once page is reloaded, DevTools will automatically reconnect.

Dismissing the dialog box, clicking on the Console menu choice and there is a ">" displayed at the top of the window.

b-morgan, 2020-02-28:
On a second system (also Windows 10, 1909, 64-bit), 1.5.79 portable does display. Graphic card is GeForce GT 640. Details are:
[NVIDIA System Information 02-28-2020 07-31-16.txt](https://github.com/balena-io/etcher/files/4267506/NVIDIA.System.Information.02-28-2020.07-31-16.txt)

Heiner1967, 2020-03-01:
same problem on my side.
But I noticed that with Version 1.5.79 1-2 month back I was able to work, now, even with the same old image it does not anymore.

Likly something around win10 has changed.

thundron, 2020-03-02:
@Heiner1967 probably :/ we have to find out what broke and how to restore it, hopefully
@b-morgan could you enable log persistence in the devtools console? it's available by clicking the cog icon, then if you refresh the window you *should* see the previous logs

b-morgan, 2020-03-02:
@thundron I followed your instructions but there was no additional information shown (i.e. same blank screens on 1.5.56, no screens at all on 1.5.79).

alexK2020, 2020-03-05:
hey, im having the same issue. when trying to run the portable i just get a white window, when running the full program theres nothing showing up. also looked at the old thread there was but there wasnt a solution or any advice. running windows 10 ltsc 1809

b-morgan, 2020-04-02:
If you are waiting on me, I just tried 1.5.79 portable to collect the requested data and it worked like it was supposed to.

syadnom, 2020-05-13:
1.5.88 nor the portable works for me.   spawns a process but GUI never shows up.

b-morgan, 2020-05-21:
This issue is intermittent. Any given version will fail on some (or all) of my Windows 10 systems and sometimes one or two systems will work correctly. Surely someone on the development team can find a Windows 10 system on which some version will fail and then get to the cause of the failure and fix it.

b-morgan, 2020-05-21:
https://github.com/balena-io/etcher/issues/3160#ref-pullrequest-622331470

lurch, 2020-05-21:
> Surely someone on the development team can find a Windows 10 system on which some version will fail and then get to the cause of the failure and fix it.

From what I understand they've been wanting to do that for ages, but none of the dev team have been able to reproduce this problem. Which is also a similar situation with #2770

b-morgan, 2020-05-27:
Apparently, the fix in https://github.com/balena-io/etcher/issues/3160#ref-pullrequest-622331470 didn't make the mainstream because https://github.com/balena-io/etcher/issues/1866#issuecomment-634516738 seems to indicate that the issue has returned in 1.5.93.

It would also be nice if the developers could tie all the issues that report the same problem together.

lmatter, 2020-05-28:
Happening on 1.5.94.  It was working on an earlier version and I upgraded.  FYI I'm on Windows Insider builds.

thundron, 2020-05-29:
@b-morgan the change *did* make it to upstream.
A GUI not showing and/or showing all grey might be due to different reasons, and the one you linked was thankfully solved.
Now, the GUI not showing and the GUI showing grey are also two *totally different issues* so I'd like to also check what the real issue is here - is the GUI showing or not?

TonyApuzzo, 2020-06-05:
Running 1.5.95 is failing to bring up the GUI for me on Windows 10 Insider 10.0.19635 Build 19635 with Nvidia GTX1080 video card (not sure if that is relevant).

Running it with the Windows Compatibility Troubleshooter and in Windows 8 mode allows it to start and run successfully.

mca-gif, 2020-06-19:
Windows 10 Pro, Insider Build 19645. I'm having the same issue with 1.5.99. It works when I enable Windows 8 Compatibility mode.

But I'm commenting to say I think there might be two separate issues with similar symptoms. In #1866 the original comment said the threads were still open and waiting. In my instance the process closes completely after a few seconds. I tried the portable version as well, but I never received a blank window like the original comment on this issue. So perhaps it is 3 separate issues.

In both the installed version and the portable version, turning on Windows 8 compatibility worked for me.

hofstee, 2020-06-19:
I'm having the same issue as @mca-gif. Windows 10 Insider 19640.

closed 2023-07-12 by dfunckt