balena-io/etcher #3089: ...Forbidden! (the bintray apt Relase file) aka E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.

dreamcat4, 2020-02-28:
Something wrong with Bintray HTTP server

```sh
sudo apt-get update
Reading package lists... Done                                                                                                                                          
E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Direct http request:

```sh
curl https://dl.bintray.com/etcher/debian/dists/stable/Release
Forbidden!
```

It also returns same `Forbidden!` response for navigating to other nearby files in same directories index / folder tree.

hollec, 2020-02-28:
Duplicate of #3078

dreamcat4, 2020-02-28:
Closed (as this is a duplicate)

closed 2020-02-28 by dreamcat4