balena-io/etcher #3090: Update error

certitudes, 2020-02-29:
I tried to update the application, but it says the public key is not available.

"Reading package lists... Done 
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://dl.google.com/linux/chrome/deb stable Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78BD65473CB3BD13
E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details."

LaurentDumont, 2020-02-29:
I think there are some issues with the repository itself.

[image]

hollec, 2020-02-29:
Duplicate of #3078

lurch, 2020-03-02:
BTW it looks like your NO_PUBKEY error is actually in relation to the http://dl.google.com/linux/chrome/deb repo. :eyes:

thundron, 2020-03-02:
Please refer to https://github.com/balena-io/etcher/issues/3078

closed 2020-03-02 by thundron