balena-io/etcher #3091: failed to burn

ParkerLouis42, 2020-02-29:
- **Etcher version:**  1.5.79
- **Operating system and architecture:** win 10 pro laptop
- **Image flashed:**  2020-02-13-raspbian-buster-full
- **Do you see any meaningful error information in the DevTools?** 

I've tried two different holders with no success.  says zOops something went wrong with the flash.  it happens right at the beginning

icanotc, 2020-02-29:
same issue but it can actually validate
but the partition wont show

lurch, 2020-03-02:
@icanotc What exactly do you mean by "the partition wont show" ? Note that Windows is unable to access Linux-format partitions.

icanotc, 2020-03-02:
@lurch its "not partitioned" under windows

lurch, 2020-03-02:
@icanotc Depending on which image you're writing, it might be a variant of this problem? https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

icanotc, 2020-03-03:
@lurch nope i tried several images and rufus never gives me a problem

lurch, 2020-03-04:
@icanotc That's because Rufus works in a different way - it formats the drive first, and then extracts the files out of the raw image onto the formatted drive. The same way as unetbootin works #413
However this approach only works with Linux images in a specific format.

Etcher works by just copying the raw image directly onto the drive - the same way as Win32DiskImager works. This approach works with **any** disk image.

@thundron Seems like this might be a good addition to the FAQ?

closed 2023-07-12 by dfunckt