balena-io/etcher #3093: Can't run balenaEtcher.Appimage

roncloneme, 2020-03-01:
I running Ubuntu 18.04.4.LTS and have tried to run the Appimge of BalenaEtcher 1.5.79.  I've made it executable.  When I try to run it I get file or directory not found.  I have had the same issue with trying to run other Appimag files as well.  I've also tried running sudo apt-get install balenaEtcher with no luck as well.
Ron

lurch, 2020-03-02:
Have you got all the necessary [dependencies](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies) installed? Are you accidentally trying to run a 32-bit AppImage on a 64-bit OS or vice-versa?

> I've also tried running sudo apt-get install balenaEtcher with no luck as well.

That's hopefully a temporary problem - see #3078

roncloneme, 2020-03-02:
I downloaded the code for the brightray.gyp file and then used nano to 
create the file and pasted in the code.  When I run the file in Ubuntu 
18.04, every line either says no such file or directory or it says 
command not found.  In doing some searching on running gyp files, it 
looked like I needed to install Node.js which I did, but still gt the 
same errors trying to run brightray.gyp.

Ron

lurch, 2020-03-02:
Sorry, I wasn't suggesting you run the brightray.gyp file, I was only pointing you at Etcher's list of dependencies.
Although I don't work on Etcher myself, so I've no idea if that list of dependencies is still up to date :man_shrugging:

roncloneme, 2020-03-02:
That's OK.  I figured I could try installing the dependencies using the 
file vice individually.  I can't seem to run any Appimage so it maybe 
some dependencies that I need to add to Ubuntu 18.04.

Thank you

Ron

closed 2023-07-12 by dfunckt