balena-io/etcher #3094: Error using debian repository

beruic, 2020-03-01:
- **Etcher version:** 1.5.76
- **Operating system and architecture:** Ubuntu 19.10 x64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** Nope

I get the following error when running an `apt update`.
```
$ sudo apt update
Ign:1 http://linux.dropbox.com/ubuntu cosmic InRelease
Get:2 http://linux.dropbox.com/ubuntu cosmic Release [6600 B]                                                          
Hit:3 http://dk.archive.ubuntu.com/ubuntu eoan InRelease                                                               
Hit:4 http://dk.archive.ubuntu.com/ubuntu eoan-updates InRelease                                                       
Hit:5 http://ppa.launchpad.net/git-core/ppa/ubuntu eoan InRelease                                                      
Ign:6 http://dl.google.com/linux/chrome/deb stable InRelease                                                           
Hit:7 http://dk.archive.ubuntu.com/ubuntu eoan-backports InRelease                                                     
Hit:9 http://dl.google.com/linux/chrome/deb stable Release                                                             
Hit:10 http://ppa.launchpad.net/hluk/copyq/ubuntu eoan InRelease                                                       
Hit:11 http://ppa.launchpad.net/peterlevi/ppa/ubuntu eoan InRelease          
Get:12 http://security.ubuntu.com/ubuntu eoan-security InRelease [97.5 kB]
Ign:8 https://dl.bintray.com/etcher/debian stable InRelease                      
Err:13 https://dl.bintray.com/etcher/debian stable Release                       
  403  Forbidden [IP: 52.28.187.224 443]
Reading package lists... Done                              
E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

whtsamri, 2020-03-01:
Duplicate of #3078

thundron, 2020-03-02:
Thanks @whtsamri , please refer to the other issue as mentioned

closed 2020-03-02 by thundron