balena-io/etcher #3095: Problems trying to flash an sd card with a .tar.gz file

DenaFrancis, 2020-03-03:
- **Etcher version:** Version 1.5.79 (1.5.79)
- **Operating system and architecture:** MacBook Pro with a macOS Mojave version 10.14.6
- **Image flashed:** ethoscope_os_20160721.tar.gz
- **Do you see any meaningful error information in the DevTools?** Tue Mar 03 2020 21:50:03 GMT+1000 (Australian Eastern Standard Time) Invalid image ({"path":"/Users/dena/Desktop/ethoscope_core_files/ethoscope_os_20160721.tar.gz","sample":0.1})

When I try to select the image ethoscope_os_20160721.tar.gz I get this message: 
Invalid Image
/Users/dena/Desktop/ethoscope_core_files/ethoscope_os_20160721.tar.gz is not a supported image type.

Balena-etcher was recommended to me by the person who created the .tar.gz file. I think that your app worked for him. 

Thanks

lurch, 2020-03-03:
I've just found and downloaded the `ethoscope_os_20160721.tar.gz` file, and it contains a whole bunch of files and directories (`/usr`, `/var`, etc.) and thus needs to be extracted to an already-partitioned and formatted disk. Etcher only works with raw disk images (typically `*.img`) where the raw disk image itself contains the partition and filesystem information.

P.S. For more about `.tar` files, see #2338

DenaFrancis, 2020-03-03:
Thank you,
It’s not a problem. There is a newer version that contains the correct file for using the galena etcher app. The developers just released it last month. 
Dena

zvin, 2020-03-07:
`Gzip`, `bzip2` or `xz` compressed __raw__ images are supported, `tar` archives are not.
The only archive file type supported is `zip` files (where etcher assumes that the disk image is the first file in the archive which name looks like an image).

closed 2020-03-07 by zvin

samaldude, 2021-11-01:
Hello I know this is an old tread and so I will give it a try. @DenaFrancis you indicated there is a newer release than ethoscope_os_20160721.tar.gz, is that correct? Would you be kind enough to direct me to it? My reference: https://zenodo.org/record/831154#.YYBLLC-cYdQ

DenaFrancis, 2021-11-01:
Hi,
Unfortunately I have not been active with the ethoscope for a while. I would check the gilestro lab website at ucl.
Thanks,
Dena