balena-io/etcher #3096: Unable to install in 18.04

meetdilip, 2020-03-03:
- **Etcher version:**  1.5.79
- **Operating system and architecture:** Ubuntu 18.04 64 bit
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** I am unable to install

I got up to `sudo apt-get update` for Debian / Ubuntu and then got an error when tried the last step
```

Get:41 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2,464 B]
Reading package lists... Done                                                  
E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
myName@my-PC-Name:~$ sudo apt-get install balena-etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package balena-etcher-electron
```

lurch, 2020-03-03:
Duplicate of #3078

nekromoff, 2020-03-05:
Same for 16.04

nekromoff, 2020-03-05:
https://dl.bintray.com/etcher/debian/etcher-electron_1.4.3_amd64.deb
Forbidden!

lurch, 2020-03-05:
See my previous comment...

nekromoff, 2020-03-05:
It was a complaint 9 days ago. If your repo is unable to be fixed in 9
days, then maybe change the repo provider.

thundron, 2020-03-05:
See #3078 as mentioned.
Meanwhile you can update/install through the release page.
Thank you for the patience

closed 2020-03-05 by thundron