balena-io/etcher #3102: Mac OS slow writing speeds

albydnc, 2020-03-10:
- **Etcher version:** 1.5.79
- **Operating system and architecture:** MacOS High Sierra 10.13.6 x64
- **Image flashed:** lubuntu iso
The writing speed over a USB2.0 drive is really slow (3 MB/s). Same setup on a linux machine gives 20-30MB/s. I think this could be caused by the dd implementation in MacOS. Using the command `dd if=/...iso of=/dev/disk5` I get 3 MB/s, while using `dd if=...iso of=/dev/rdisk5` gives me full speed.

lurch, 2020-03-10:
See https://www.google.com/search?q=%2Fdev%2Fdisk+vs+%2Fdev%2Frdisk for more info.

thundron, 2020-04-28:
We don't use `dd` but we do write byte-to-byte and there are other reports of slow performance on OSX High Sierra, some solved by updating as seen [here](https://github.com/balena-io/etcher/issues/2770#issuecomment-619013728), can you please try updating to Mojave (and above) and report back?

JustJoeTX, 2020-07-16:
I'm running in to the same issue with Catalina 10.15.5. 
Writing to a 128GB Micro SD, the disk speed starts out good then suddenly drops to 0.3MB/s and the time to complete extends from 160 minutes (which already seems long) to over 5,000 minutes to complete.

lurch, 2020-07-17:
@JustJoeTX That sounds like it might be a hardware problem with your SD card and/or SD card-reader? Might be worth trying with a different SD card and/or different flashing software? :shrug:

thundron, 2020-07-21:
As @lurch pointed out, it's likely something to do with the flashing media rather than Etcher. Please try with both another sd-card and another card reader.
For what it's worth, it started to happen with a drive I've been using too for years now. It's often due to the target reaching it's EOL, be it for extensive usage or defective quality (in case it's new)

closed 2020-07-21 by thundron