balena-io/etcher #3103: Rate and ETA font too small on Mac

GaryBoone, 2020-03-11:
- **Etcher version:** 
1.5.79

- **Operating system and architecture:** 
MacOs Catalina (10.15.3)

While flashing, there's an orange progress bar on the right. Underneath it is some tiny text with the rate and the ETA. That's useful to see but is way too small to read clearly.

Here's an image showing the relative sizes. On my screen, the height of the progress bar is about the width of a pencil. The text below might be a quarter the height of the mouse cursor arrow.

[image]

EstevaoCostaG3, 2020-05-18:
Same issue. But to me the font size of the whole app is too small. It's really hard to read those tiny texts for me. To give you an idea, I almost always have to zoom in around 125-150% to be able to read things on the internet (right now on github the zoom is actually set to 160%).
- My suggestion: the app could provide an option to change the font size in the settings menu.
Here's a screenshot of the app running on my machine, it has a 15.6" monitor.
[image]

- Etcher version: 1.5.79
- OS: Ubuntu 20.04 LTS

lurch, 2020-05-18:
> I almost always have to zoom in around 125-150% to be able to read things on the internet (right now on github the zoom is actually set to 160%).

Wouldn't it just be easier to set the global zoom level in your OS's preferences?