balena-io/etcher #3107: macOS Catalina authentication

seamusdemora, 2020-03-16:
I operate as a NON-ADMIN user on my Macs. Etcher prompted me for a PASSWORD before flashing a USB. I tried entering the ADMIN password first, but it REJECTED that. I then entered my NON-ADMIN password. Etcher dod nothing... it just sat there & looked back at me :) 

Took the same ISO over to my macOS 10.14.6, Etcher prompts for BOTH username and password, then proceeds to flash correctly.

wtf, o??

- **Etcher version:** 
Version 1.5.79 (1.5.79)
- **Operating system and architecture:** 
macOS 10.15.3 
 Model Name:	MacBook Pro
  Model Identifier:	MacBookPro16,1
  Processor Name:	8-Core Intel Core i9
  Processor Speed:	2.3 GHz
  Number of Processors:	1
  Total Number of Cores:	8
  L2 Cache (per Core):	256 KB
  L3 Cache:	16 MB
  Hyper-Threading Technology:	Enabled
  Memory:	64 GB

- **Image flashed:** 
linuxmint-19.3-cinnamon-64bit.iso

- **Do you see any meaningful error information in the DevTools?** 

NO

lurch, 2020-03-17:
More info in #2997 and #2833

seamusdemora, 2020-03-20:
Thanks - Got it, but... I'm using 1.5.79, and it *still* doesn't work on Catalina.

lurch, 2020-03-21:
First of all: I don't work on Etcher, and I don't have a Mac, so these are largely all guesses.... (but maybe theses guesses will be helpful anyway? :confused: )

From what I've been reading in various Etcher issues, it sounds like MacOS has both "admin accounts" and "non admin accounts"? In versions of MacOS prior to Catalina, Etcher used the "system elevation mechanism" (via @jorangreef 's  [sudo-prompt](https://github.com/jorangreef/sudo-prompt) ) which then prompts for both a username and password, and people running as a non-admin user would then enter their admin-username and admin-password into the prompt, and Etcher would then gain full access.
However in MacOS Catalina this "system elevation mechanism" (?) stopped working, and so Etcher was changed to instead display a custom password-prompt, which then runs sudo as the currently-running user ([this comment](https://github.com/balena-io/etcher/issues/2997#issuecomment-565794526) shows what the different password prompts look like). I guess this works fine for everyone who's running as an admin-user, but doesn't work for people like @seamusdemora who are running on Catalina as a non-admin user? And I also guess that @zvin and @thundron are always running as admin-users, which is maybe why they never saw this issue themselves?

AFAICT these are all "security restrictions" introduced in MacOS Catalina by Apple, so there's not really anything that Etcher can do to work around them?  :man_shrugging: 
[This comment](https://github.com/jorangreef/sudo-prompt/issues/109#issuecomment-549782546) says that even `PrivilegedHelperTools` (whatever they are) don't work in Catalina either.

rradar, 2020-03-21:
@seamusdemora their is this little [tool here](https://gitlab.com/bztsrc/usbimager/raw/binaries/usbimager_1.0.0-intel-macosx-cocoa.zip) and it works like a charm for me. It also supports your macOS 10.14. Maybe that could be a shortcut for you.

seamusdemora, 2020-03-21:
@lurch  - Thanks, this is great feedback. All of your 'guesses' sound correct AFAICT (I'm no expert on macOS security mechanisms beyond the fact that they're a PITA). Good insights for someone that doesn't use a mac! 

So - a couple of things: 

1. ***IF*** Etcher uses `sudo` in Catalina for privilege (do we know this - or not?), then this becomes an [easy fix!](https://apple.stackexchange.com/questions/360198/allow-non-admin-user-to-sudo-without-access-to-admin-account-data/360272#360272) ([alternatively](https://github.com/seamusdemora/seamusdemora.github.io/blob/master/UsingSudoInMacos.md)) I say *"easy"* because I also assume we can learn what **specific** command is used by Etcher to enable flashing. 

2. I can't agree that there's nothing Etcher can do to *work around* Catalina's security restrictions. If that were true, how have so many other apps managed to clear the bar? PITA? - Yes, definitely!; Can't be done? - Don't think so!

rradar, 2020-03-22:
[image]

lot's of censorship recently...

zvin, 2020-03-22:
@seamusdemora we use `sudo` on catalina, you can check the code here https://github.com/balena-io/etcher/blob/master/lib/shared/catalina-sudo/sudo.ts#L41 .

lurch, 2020-03-22:
> If that were true, how have so many other apps managed to clear the bar?

@seamusdemora Like I said in my previous message - I was only guessing, and I've never used MacOS :green_apple:  :man_shrugging:  I never claimed any of my guesses were 100% correct :laughing: 
But note that Etcher requires raw block-level disk access, which is a different operation (and so _may_ have different security restrictions?) than simply editing a root-owned text file :lock:  :man_shrugging:

seamusdemora, 2020-03-22:
@zvin - I don't see anything there that helps, but . If you're familiar with `sudo` you know that it can allow an unprivileged user to execute commands reserved for privileged users. There are in general 2 ways to do this: give the unprivileged user access to all privileged commands, or (as is typical) designate **specifically** what commands are allowed. I'm completely ignorant wrt javascript, and I can see nothing that looks like a _sudo-able_ command. I suppose I was thinking you would use a `mount` command in etcher to mount the thumb drive before writing. I guess that's not the case?

I'd try adding `etcher` to my `sudo` list, **_but_** it's a big taboo to add a GUI app to `sudo` for reasons I cannot explain precisely.

zvin, 2020-03-23:
@seamusdemora Etcher does not mount anything. It writes disk image to the drive directly.
It's an electron app, so the only thing being executed is electron.
The process that writes to the drive is not the whole app, it is [electron running as node](https://www.electronjs.org/docs/api/environment-variables#electron_run_as_node) so no ui is being ran with `sudo`. Only [this](https://github.com/balena-io/etcher/blob/master/lib/gui/modules/child-writer.ts) runs with `sudo`.

It looks like there is a way to install a privileged helper according to https://www.tweaking4all.com/software/macosx-software/applepi-baker-v2/#WritingtoaDiskSudoAccessvsHelperTool . I'm not sure the privileged helper may be electron. I don't have time to work on this for now. Pull requests are welcome.

lurch, 2020-03-23:
@seamusdemora I think the articles you mentioned are only about _fine-tuning_ what permissions `sudo` has access to. AIUI Etcher is currently using the "blanket permission for everything" approach to `sudo`. So I may be wrong, but if the "blanket permissions" approach doesn't work for running Etcher as a non-admin user on MacOS Catalina, then I don't think _fine-tuning_ those permissions will get you anything extra? Although I've never dived down the sudo-configuration rabbit-hole, so there may be parts I'm missing or not understanding :wink: 

@zvin The author of ApplePiBaker has also written about the problems he's been having with Catalina in [this forum thread](https://forums.developer.apple.com/thread/124185) - I've no idea if he ever came up with a solution though? :man_shrugging:

seamusdemora, 2020-03-25:
You're probably correct. So many crappy things with Catalina, and I'm tired of fighting them all. I'm putting this issue into a dirty bag labeled "Catalina Crap", and I'll come back to it.

seamusdemora, 2020-04-19:
@zvin  -  Finally got around to reading your last post (March 23). That's interesting, but unfortunately I know nothing about Electron.

With respect to the *privileged helper*, I don't know much about that either, BUT I do know that [`LaunchControl`](https://www.soma-zone.com/LaunchControl/) uses a *privileged helper* to good effect. None of my *pre-Catalina* `launchd` jobs would run under Catalina without it. So - the developer of `LaunchControl` has (with some misgivings I imagine) gone along with Apple's _diktat_.

The post you linked to re apple-pi baker is, for me, very depressing. I say that because Apple's new "security measures" seem designed to benefit Apple - not their customers! The entire business of "helper apps" is horrible and disrespectful. Apple has imagined their customer base is 100% teenaged girls with popcorn for brains, and they are hell-bent on sowing the seeds of their own demise - just as they have done previously... ah! sorry - please excuse the rant... now where was I?

Oh yes - Etcher can no longer write to a USB device under macOS Catalina... You said you didn't have time for this now, and I understand that completely. But if you read through the post re Apple-Pi-Baker, you will know that this "helper app" business can only be done by you - the helper app and the app itself must both be "signed" by you. 

If I were in your place, I would be very tempted to say, "No - I refuse to allow Apple to subjugate me." Personally, and in spite of how much I like Etcher, I hope this is what you do. I say this because I believe that if all 3rd party developers told Apple to GFY, Apple would suffer financially. And it's clear that the best way to get an accountant CEO's attention is to put a hit on revenue. Apple's stockholders will give Tim Cook the same treatment they gave John Sculley (Pepsi-man) years ago.

lurch, 2020-04-20:
> I say this because I believe that if all 3rd party developers told Apple to GFY, Apple would suffer financially.

Unfortunately I believe those 3rd party developers would probably suffer more financially than Apple would... :confused:

seamusdemora, 2020-04-21:
>... 3rd party developers would probably suffer more financially than Apple

Sadly, you are probably correct. But broken software doesn't generate much revenue either - and where does it end?

seamusdemora, 2020-04-26:
I started etcher yesterday, and during the [upgrade](https://github.com/balena-io/etcher/blob/master/CHANGELOG.md), macOS flashed a notice re a new **helper app** being installed. Alas, this isn't the **helper app** needed that allows etcher to work for non-admin users on Catalina.

seamusdemora, 2020-04-30:
[This post was called to my attention](https://github.com/jorangreef/sudo-prompt/issues/109#issuecomment-548851079). This is a potential solution.

lurch, 2020-04-30:
That's a link to a comment by @zvin , who is one of the Etcher developers? :confused:

seamusdemora, 2020-04-30:
@lurch : ha ha... well I'm sure he'll appreciate that  :) 

Maybe I'm too anxious... I'm going to go back to `rufus` on my old Windows 7 box for a while. I'm sure this will be fixed at some point.

lurch, 2020-05-01:
Until this _does_ get fixed, it's probably worth leaving this issue open, rather than closing it? :slightly_smiling_face:

macarena, 2020-06-11:
hey guys, sorry if I'm commenting at the wrong issue or if someone else already comments this and I didn't read. But I think this is related to the new permission system. It worked for me when I gave the etcher total access to the disks. What I have seen in other apps that etcher don't have is a pop-up window explaining this to the user, so I also suggest it (assuming this is the problem/fix, of course).

[image]

seamusdemora, 2020-09-16:
I opened this issue, then closed it because I decided to quit using Etcher ('cause didn't work properly on Catalina), then re-opened it at the request of another user. I stopped using Etcher as it didn't seem to be well-supported. **_I can't tell from this thread whether anything has been fixed or not????_** 

I'm closing the thread (again) because I still get notification of posts here. For those still struggling with Etcher's authentication issue on Catalina, I'd suggest it's time to start a new issue.

seamusdemora, 2022-05-29:
I don't know when - or how - this happened, but Etcher now works on my macOS Catalina system. The "About" window indicates the version is 1.7.9

closed 2022-05-29 by seamusdemora