balena-io/etcher #3108: Etcher 1.5.79 Failed to write to USB Kingston nanolinux64-1.3

PhilHNewman, 2020-03-24:
Error when writng .iso to Flash drive

- **Etcher Version: 1.5.79** 
- OS Name	Microsoft Windows 10 Home
Version	10.0.18363 Build 18363
OS Manufacturer	Microsoft Corporation
System Manufacturer	LENOVO
System Model	20404
System Type	x64-based PC
System SKU	LENOVO_MT_20404_BU_idea_FM_Lenovo Flex 2-14
Processor	Intel(R) Core(TM) i3-4010U CPU @ 1.70GHz, 1701 Mhz, 2 Core(s), 4 Logical BIOS Version/Date	LENOVO A0CN27WW, 29/07/2014
SMBIOS Version	2.7
Embedded Controller Version	1.20
BIOS Mode	UEFI
 
- **Image flashed: nanolinux64-1.3** 
- **Do you see any meaningful error information in the DevTools?** 

Trim: false
C:\Users\Phil\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Tue Mar 24 2020 17:34:13 GMT+0000 (Greenwich Mean Time) Flash error ({"image":"C:\\Users\\Phil\\Downloads\\nanolinux64-1.3.iso","drives":[{"size":30943995904,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"F:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"F:\\"}],"description":"Kingston DataTraveler 3.0 USB Device","isUAS":false}],"driveCount":1,"uuid":"c66745e8-007e-4316-93d7-d9ea586e4b01","flashInstanceUuid":"c66745e8-007e-4316-93d7-d9ea586e4b01","unmountOnSuccess":true,"validateWriteOnSuccess":true,"trim":false,"applicationSessionUuid":"42b848a7-4be3-4ec0-aa0b-9fc7f16569e8","flashingWorkflowUuid":"44e418e6-b21c-4e83-b4be-66e9eff04175","error":{"name":"Error","message":"Couldn't clean the drive, Command failed: diskpart s n (code 2147942405)","stack":"Error: Couldn't clean the drive, Command failed: diskpart s n (code 2147942405)\n    at n    at Generator.throw (<anonymous>)\n    at rejected (C:\\Users\\Phil\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\etcher-sdk\\build\\diskpart.js:21:65)"},"sample":0.1})
C:\Users\Phil\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Finish: 0
C:\Users\Phil\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Terminating IPC server
C:\Users\Phil\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Flash results {results: {…}, cancelled: false}
C:\Users\Phil\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Tue Mar 24 2020 17:34:16 GMT+0000 (Greenwich Mean Time) Write failed ({"image":"C:\\Users\\Phil\\Downloads\\nanolinux64-1.3.iso","drives":[{"size":30943995904,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"F:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"F:\\"}],"description":"Kingston DataTraveler 3.0 USB Device","isUAS":false}],"driveCount":1,"uuid":"c66745e8-007e-4316-93d7-d9ea586e4b01","status":"failed","flashInstanceUuid":"c66745e8-007e-4316-93d7-d9ea586e4b01","unmountOnSuccess":true,"validateWriteOnSuccess":true,"trim":false,"applicationSessionUuid":"42b848a7-4be3-4ec0-aa0b-9fc7f16569e8","flashingWorkflowUuid":"44e418e6-b21c-4e83-b4be-66e9eff04175","error":{"description":"Please try again, and contact the Etcher team if the problem persists","report":false,"code":"ECHILDDIED"},"sample":0.1})
/C:/Users/Phil/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png:1 GET file:///C:/Users/Phil/AppData/Local/Programs/balena-etcher/resources/app.asar/lib/assets/icon.png net::ERR_FILE_NOT_FOUND

lurch, 2020-03-25:
Duplicate of #2812 ?

PhilHNewman, 2020-03-25:
The issue is not with etcher. The usb becomes inaccessible when the iso is written. Unable to use diskpart to access usb

Regards

Phil Newman

closed 2023-07-12 by dfunckt