balena-io/etcher #3111: Home Page MacOS download link unresponsive

dan003400, 2020-03-26:
The homepage download link is not responding for MacOS on the top Hero.

mrkapqa, 2020-03-26:
same goes for Linux Version

mrkapqa, 2020-03-26:
downloads can be accessed however from here

https://github.com/balena-io/etcher/releases/

lurch, 2020-03-26:
@zvin @thundron I'm seeing the same :confused: 
Clicking on the button at the top:
[image]
doesn't do anything.
However the download links further down the page:
[image]
**do** all work correctly.
(i.e. it seems the download links are all okay, it's just the top-button which isn't working properly)

zvin, 2020-03-26:
Thanks @lurch , it will be fixed quickly.

lurch, 2020-03-26:
@zvin This is working (for me) again now :+1:

closed 2020-03-27 by zvin