balena-io/etcher #3116: Strange functionality after running etcher

aawheatley, 2020-04-08:
- **Etcher version:** 1.5.79
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** none
- **Do you see any meaningful error information in the DevTools?** no

**What is the actual issue?**
After I downloaded the tool and executed it, something strange happens. You can see the video to see what exactly is the problem.
[https://youtu.be/f6cDaekuSWM](https://youtu.be/f6cDaekuSWM)

thundron, 2020-04-08:
Where did you download it from?
Regardless, try downloading it from our website https://balena.io/etcher or the releases page here on github (https://github.com/balena-io/etcher/releases)

aawheatley, 2020-04-08:
Hi,
The download came from TOR Project Tails download page (https://tails.boum.org/etcher/Etcher-Portable.exe) Also, I believe it may have been tampered with because when it flashes an image, it doesn't work properly. Doesn't finish doing what it is supposed to and always exits with an error that the checksum is not equal.

aawheatley, 2020-04-08:
After downloading and installing from the official page, check this out: [https://youtu.be/dRqzAkZoKos](https://youtu.be/dRqzAkZoKos)

aawheatley, 2020-04-08:
This time, after using the official download, it finished with no errors. Glad it worked but it still is strange for the calculator to be opening.
[image]
1

lurch, 2020-04-08:
Have you got an up-to-date virus checker installed? In the second video, are you sure you're launching the "official Etcher" and not accidentally still running the Etcher from your first video?

aawheatley, 2020-04-08:
Yes, my antivirus stays up to date as per a rule I configured and in the second video, I deleted the old etcher and installed the new one from the official site.
The one that I initially used in the first video seemed to be a portable version.

lurch, 2020-04-08:
Portable versions can also be found at https://github.com/balena-io/etcher/releases if you want to test the "official portable version" for comparison?

aawheatley, 2020-04-11:
This issue can be closed. Don't know where the problem lays but I think it may something with my OS. Been having some issues with it lately and will reinstall it.
Thanks for the help!

closed 2020-04-24 by aawheatley