balena-io/etcher #3118: Will Etcher ever support multiple partitions?

TheOnlyArtz, 2020-04-09:
Hey there! I've been looking around for a software which can flash ISOs to a drive with multiple partitons, for example I have a single 512GB hard drive with 2 partitions on it, and using Etcher I can't flash an iso on a single partition out of the two
[image]

lurch, 2020-04-09:
Duplicate of #2670  #2459  #1524 #1317  #453 etc.
Those issues explain why Etcher doesn't (can't?) support what you're asking for.

#414 and #413 mention some alternatives which _might_ do what you want...

raybellis, 2020-07-31:
I believe whoever is closing this doesn't understand the request.

All that most people need is the ability to write an image of a complete file system into a specific _existing_ partition. 

On most OS's those partitions will already be presented to the OS as a separate raw block device per partition,

-  on macOS you get `/dev/disk<n>s<p>` (and something similar on Linux)
-  on Windows you get `\\?\Device\Harddisk<n>\Partition<p>`

In theory all this needs is a selector that enumerates the possible destinations in that format, and then the OS will take care of everything else for you, including ensuring that you can't stomp over the start of the next partition.

See also [dd.exe](http://www.chrysocome.net/dd)

lurch, 2020-07-31:
I don't work on Etcher, but just to add my 2 cents...
A lot of the time, people are using Etcher to write Raspberry Pi IMG files (including e.g. [balenaOS](https://www.balena.io/os/) ), and these almost always contain multiple partitions and their own partition table; which wouldn't work when "just written to a single partition".

On a side note, I believe that just writing an ISO to an arbitrary partition doesn't guarantee that your BIOS will be able to boot from that partition?

And then of course there's "fun" things like https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

But meh, I'm just a random internet person like everybody else :wink: 

> (and something similar on Linux)

The Linux equivalent would be something like `/dev/sdb3` which would be the third primary partition on `/dev/sdb` (with `/dev/sdb` typically being the second drive in the system - gets slightly more complex when IDE or NVME is involved).

raybellis, 2020-08-01:
On 31/07/2020 20:00, Andrew Scheller wrote:
> I don't work on Etcher, but just to add my 2 cents...
> A lot of the time, people are using Etcher to write Raspberry Pi IMG
> files (including e.g. balenaOS <https://www.balena.io/os/> ), and these [...]

And Etcher is fine indeed for that.  By all means write those to a whole
card volume.

> On a side note, I believe that just writing an ISO to an arbitrary
> partition doesn't guarantee that your BIOS will be able to boot from
> that partition?

There are other uses for writing ISOs to partions than booting from them.

In particular it's *incredibly* useful for use with scsi2sd, which can
present individual partitions as separate logical SCSI drives to a
sampling synthesizer.

lurch, 2020-08-01:
> There are other uses for writing ISOs to partions than booting from them.

I have no evidence to back this up, but I _suspect_ that most people who want to write ISOs to an individual partition want to be able to write multiple Live-Linux ISOs to separate partitions on a e.g. a large USB thumb-drive so that they have a single thumb-drive that they can boot multiple separate OSes from? (but I think that this requires some kind of intermediate bootloader, so isn't as simple as people think it is)

> In particular it's *incredibly* useful for use with scsi2sd

Maybe that's a bit of a niche use-case, below the Etcher-developers' radar? :shrug:

raybellis, 2020-08-02:
On 01/08/2020 14:13, Andrew Scheller wrote:

>     In particular it's /incredibly/ useful for use with scsi2sd
> 
> Maybe that's a bit of a niche use-case, below the Etcher-developers'
> radar? 🤷

Whatever, it's a valid use case, and in theory it's trivial to support.

I'm on FB groups where there's loads of people crying out for a decent
solution to being able to write an ISO full of samples to a specific
partition within an SD card.

TheOnlyArtz, 2020-08-14:
Reopening the issue since there's a healthy conversation about it.

pryorda, 2020-08-14:
Today I was trying to do a kali install with a peristent volume, but etcher was formatting the whole disk

rgetz, 2021-01-06:
A more concrete (but niche) use case:

When a Intel (Altera) SoC boots, the first thing it does, is load a "pre-loader" from a specific partition on a SD Card. Their developer tools create a `preloader-mkpimage.bin` file, which end users have to install into a specific partition on an SD Card. (Intel expects every user to be a developer) This normally isn't a problem for developers, as Intel makes a command line tool called `alt-boot-disk-util` which [does this](https://rocketboards.org/foswiki/Documentation/SoCSWWS1IntroToAlteraSoCDevicesLab1Preloader#Using_the_34alt_45boot_45dis_AN1), but it's part of their giant [SoC EDS](https://www.intel.com/content/www/us/en/software/programmable/soc-eds/overview.html) package, which is licensed, and large.

This brings me to my problem - I create a standard userspace image that runs on Pi, Xilinx SoCs (Zynq & Zynq Ultrascale), and Intel SoCs (Stratix, Arria and Cyclone). This weird partition thing that Intel does, makes it impossible to make a generic SD image, that can be booted on different hardware platforms without telling end users they need to re-image specific partition on their SD card. I have an SD image (an `*.img`), which I tell end users to write to the entire card (which leaves the special Intel partition blank), and I would like to tell them they can configure things for their specific Intel SoC, by using etcher (since it's small, cross platform, and end user friendly), but right now I can't...

It's still a byte-by-byte copy, it's just starting at a the start of partition, not at block 0.

There are lots of "embedded" use cases, where you need to be able to pick a file, and write a partition, not the entire drive. There are too many users which are not familiar enough with command lines, and I'm always a little afraid of telling people to use `dd` and having someone wipe out their hard drive by accident...

There is already "unsafe" mode, there could be a "expert" or "embedded" mode as well.

thundron, 2021-02-02:
it's already been partially discussed in https://github.com/balena-io/etcher/issues/414#issuecomment-219723312 - it's outside of the scope of Etcher, but we were considering the addition of a plugin system in the future

closed 2021-02-02 by thundron

lurch, 2021-02-03:
@thundron When you say "but we were considering the addition of a plugin system in the future", do you mean "we were considering it (but we're not any more)" or "we are (still) considering adding it in future" ?

thundron, 2021-02-03:
@lurch it hasn't been discussed recently so I wouldn't know about the present since it isn't a short task

lurch, 2021-02-03:
Ahh, so "we were considering it, and we (might) consider it again in future" :grinning:

brokenmold, 2021-03-15:
The wide use-case for this would be for Live Persistence on certain linux distributions.
ie: Kali, Ubuntu, etc..

I praise etcher for being OS-Agnostic and having a super clean UI.
However, It's simply a non-starter if you need persistence for what you're flashing.

lurch, 2021-03-15:
@brokenmold see https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#how-can-i-configure-persistent-storage

nwetter2, 2022-07-29:
> it's already been partially discussed in [#414 (comment)](https://github.com/balena-io/etcher/issues/414#issuecomment-219723312) - it's outside of the scope of Etcher, but we were considering the addition of a plugin system in the future

Your reference to 414 makes me think you don't understand the request. 414 seems to be requesting etcher "make something bootable", which is clearly beyond the scope of a tool intended to write raw images without modification. All I need to do is overwrite an existing partition with a raw image and verify bit-perfect accuracy. As explained in this thread it appears trivial to implement (I do it myself in Linux but would like a Windows tool), and it's not obvious to me why it should be out of scope.

lurch, 2022-07-29:
Again, just my opinions as an outside observer, I'm nothing to do with Etcher or Balena...

> All I need to do is overwrite an existing partition with a raw image and verify bit-perfect accuracy. ... it's not obvious to me why it should be out of scope.

Because that's an advanced use-case, and the more options Etcher adds, the easier it becomes for an "average" user to screw up their system? :shrug: I believe that https://github.com/balena-io-modules/etcher-sdk was created to make it easier for people to support their own advanced use-cases?

rgetz, 2022-07-29:
I guess - I personally don't need it anymore  - we decided to extend https://github.com/raspberrypi/rpi-imager to do what we needed. Thanks.

dandv, 2023-04-19:
My use case is very simple and common, and I'm surprised it's not mentioned - large drives.

Pretty much every single modern flash drive is much larger than ISO images. Currently Etcher wastes all that extra space.

I have a very nice and compact Samsung Fit 128GB drive with my files, and I'd like to make it bootable as well. That's impossible with Etcher.

lurch, 2023-04-19:
> The Linux images it flashes are read-only. I don't even need multiple partitions, just let me write on the bootable one. (Maybe this is how these ISOs are supposed to operate, but as an end user, that's an annoying limitation.)

https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#how-can-i-configure-persistent-storage