balena-io/etcher #3119: libpango1.0-0 dependency not available in Debian Bullseye

arrowmaster, 2020-04-13:
- **Etcher version:** 
1.5.80
- **Operating system and architecture:** 
Debian Bullseye amd64
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

The transitional package libpango1.0-0 has been removed from Debian Bullseye. The package dependency needs to be changed to `libpango-1.0-0` or `libpango-1.0-0 | libpango1.0-0`.

Thoulah, 2020-04-22:
```
~$ sudo apt dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gconf-service gconf2 gconf2-common libappindicator1 libdbusmenu-gtk4 libgconf-2-4 libindicator7 libpangox-1.0-0
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  balena-etcher-electron libpango1.0-0
The following packages will be upgraded:
  gir1.2-pango-1.0 libpango-1.0-0 libpango-1.0-0:i386 libpangocairo-1.0-0 libpangocairo-1.0-0:i386 libpangoft2-1.0-0 libpangoft2-1.0-0:i386 libpangoxft-1.0-0
8 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
Need to get 683 kB of archives.
After this operation, 283 MB disk space will be freed.
```

Please fix the dependencies. Thank you!

lurch, 2020-04-23:
Had a quick browse, and the reference to libpango seems to be [here](https://github.com/balena-io/etcher/blob/master/electron-builder.yml#L94)

carlosnewmusic, 2020-04-28:
[image]

alu18999959g, 2020-05-02:
I just fixed it but had to rebuild the deb package.

- Download the .deb installer file

- Inside the folder your downloaded the file run dpkg-deb -R ./balena-etcher-electron_1.5.80_amd64.deb balena-etcher-electron

- Edit file balena-etcher-electron/DEBIAN/control and replace libpango1.0-0 with libpango-1.0-0

- Rebuild .deb installer running dpkg-deb -b balena-etcher-electron balena-etcher-electron_1.5.80-fixed_amd64.deb

- Install  balena-etcher-electron  using the fixed installer running sudo dpkg balena-etcher-electron_1.5.80-fixed_amd64.deb

`sudo dpkg -i balena-etcher-electron_1.5.80-fixed_amd64.deb 
Seleccionando el paquete balena-etcher-electron previamente no seleccionado.
(Leyendo la base de datos ... 508614 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar balena-etcher-electron_1.5.80-fixed_amd64.deb ...
Desempaquetando balena-etcher-electron (1.5.80) ...
Configurando balena-etcher-electron (1.5.80) ...
Procesando disparadores para desktop-file-utils (0.24-1) ...
Procesando disparadores para mime-support (3.64) ...
Procesando disparadores para hicolor-icon-theme (0.17-2) ...
`
[image]

[image]

**Final steps**

`cd /opt/balenaEtcher`
`sudo chown root:$USER chrome-sandbox`
`sudo chmod 4755 chrome-sandbox`
[image]
[image]
[image]

lurch, 2020-05-04:
> cd /opt/balenaEtcher
sudo chown root:$USER chrome-sandbox
sudo chmod 4755 chrome-sandbox

Is that the fix for #3129 ? :shrug:

zvin, 2020-05-05:
fixed in v1.5.84

closed 2020-05-05 by zvin