balena-io/etcher #3124: Etcher broke my USB

Andi7891, 2020-04-22:
Etcher version: 1.5.76
Platform: Windows 10 64 bits
Image: Debian for raspberry pi

I tried to flash the latest Raspbian image, but in the end he told me 1 failed device, when i tried to format he gave me this:
[image]
[image]
I also tried linux to format it.

lurch, 2020-04-22:
Do any of the comments / suggestions in #2102 help?

Andi7891, 2020-04-22:
I tried all

thundron, 2020-04-24:
Your device either reached it's EOL or was defective in the first place.
Etcher doesn't (and can't, actually) compromise hardware in any way.
Please see [our docs](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows) for some solutions available on different OSes (which I suggest trying, particularly a Linux one).
For a deeper explanation on the subject, there's also a nice post by Rufus [here](https://github.com/pbatard/rufus/wiki/FAQ#help-rufus-damaged-my-flash-drive)

closed 2020-04-24 by thundron

Andi7891, 2020-04-24:
I used sd card before with 3 minutes and he worked

lurch, 2020-04-24:
@Andi7891 Possibly you've got a "fake" SD card - I've never had one myself, but from what I understand they work when you try writing a small amount of data to them, but as soon as you try writing a "big" amount of data to them (whether that's using Etcher or Windows Explorer) they then break. You should be able to return your SD card under warranty for a replacement.

Andi7891, 2020-04-24:
No more warranty and he is hama from a official market