balena-io/etcher #3125: Portble Version Doesn't Seem to be Portable

teamayu, 2020-04-22:
The portable version doesn't seem to be portable:
AppData/Roaming/balena-etcher/*

Loads of files/folders in the AppData folder. How do I make the portable version portable?

lurch, 2020-04-23:
Duplicate of #2871 ?

thundron, 2020-04-24:
Unfortunately, that's how the packager works for now, it just uses standard folders.
Please see the other issue for more infos, closing as duplicate

closed 2020-04-24 by thundron

chocmake, 2023-02-26:
> Duplicate of #2871 ?

That issue currently 404s. Found it also writes to the registry: `HKEY_CLASSES_ROOT\etcher` where it contains the path to the copy of the executable that is placed in the %temp% directory.

Perhaps could have a note/tooltip icon beside the portable download to clarify it's not portable in the typical Windows sense (ie: writing data only to the the same, user-configured directory that the executable is contained within, such as on a USB drive, and not elsewhere on the system).

Mansarde, 2023-04-22:
> Unfortunately, that's how the packager works for now, it just uses standard folders. Please see the other issue for more infos, closing as duplicate

Can you tell us the title of the issue? Because as @chocmake said, the link/issue is not available.
Thanks in advance!