balena-io/etcher #3128: GSOD (Green Screen Of Death, for Windows Insiders) when flashing an image.

WG-MakerOfThings, 2020-04-24:
- Etcher version: v1.5.81 
- Operating system and architecture:  Windows 10 Home Insider Preview 19613.1000
- Image flashed: raspbian-buster-lite.img  
- Do you see any meaningful error information in the DevTools? No.

zvin, 2020-05-04:
This is most probably an issue with your hardware or drivers. Please try flashing with other software (rufus for example), you'll probably get the same BSOD.

closed 2020-05-04 by zvin

WG-MakerOfThings, 2020-05-07:
Hello,
I have tried some other imagers (Win32DiskImager and the Raspberry Pi
Imager), but have not got any BSODs. The blue screen happened when I was
flashing Raspbain to a MicroSD card. When I got the blue screen,
BalenaEtcher was 5% done verifying the image that I was in the prosses of
flashing.

lurch, 2020-05-07:
Was that a one-off, or does it happen every time when flashing that image with Etcher? Does it always fail in the same place?

zvin, 2020-05-07:
There is nothing Etcher can do about a BSOD.
Etcher runs in user space.
https://stackoverflow.com/questions/7034592/create-bsod-from-user-mode

WG-MakerOfThings, 2020-05-12:
> Was that a one-off, or does it happen every time when flashing that image with Etcher? Does it always fail in the same place?

Yes. Etcher starts flashing an image, finishes successfully, and then starts verifying. At around 5%, Etcher freezes, and then I have a BSOD.
When I use some other apps, they finish successfully without a BSOD.

thundron, 2020-05-13:
@WG-MakerOfThings do the other tools perform the verification too?
anyway, it's likely an issue with hardware/drivers, you should see the message code in the BSOD that tells you more about the issue.
other than that, there's nothing we can do - a BSOD is a windows issue at managing *something*, etcher just happens to be an application using that *something* that breaks (again, the BSOD surely contains more information)