balena-io/etcher #3129: The SUID sandbox helper binary was found, but is not configured correctly.

Hierosme, 2020-04-25:
- **Etcher version:** 
balenaEtcher-1.5.81-x64

- **Operating system and architecture:** 
Debian 10.3 - 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux

- **Image flashed:** 
debian-10.2.0-arm64-netinst.iso

- **Do you see any meaningful error information in the DevTools?** 
The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenatOjWuh/chrome-sandbox is owned by root and has mode 4755.

- **My repport issue**
Now it do years, where Etcher is unusable for users. Etcher have no interest if users can't do anything with it. I understand the need of root acces, but by the paste etcher haven't require root acces. Please back to a situation where users can use it by default or please display a message error.

Actually it take 100Mo to download it and 10 minutes for understand a workarround. The end result is very bad because use "dd" is more simple, take less time, and work ...

It really kill the reputation of etcher, "no" it not simple anymore, "no" it's not work out of the box anymore.

I hope users feed back will help.

Regards

lurch, 2020-04-25:
I wonder if this might have anything to do with #2569 ? :man_shrugging:

Hierosme, 2020-04-25:
For me it have no relation ...

My point is to display a message for inform users about a trouble.

#2569 look to be hardning setting where /tmp have noexec option.

DougBeney, 2020-05-07:
It successfully runs if you run it as sudo.

Hierosme, 2020-05-08:
Hello, 

yes it work when run by sudo, unfortunaly  sudo is not a command install by default on a Debian system, and sudo require setting. (/etc/sudoers)

My point is to simply inform the user with a message.

And by the way permit **Users** to use it own **gvfs** media for do what the user want to do.

If i need **sudo**, i not need **balenaEtcher** (150Mo), because *dd* is more simple.

Actually use **balenaEtcher** require more work , compare to use the normal way **sudo** with  **dd**

By the paste **balenaEtcher** have never require **sudo** for start.
if i need **sudo** just for start the application . why use **balenaEtcher** (150Mo) ? where is the gain ?

Hierosme, 2020-05-08:
> It successfully runs if you run it as sudo.

That is totaly not my point.

DougBeney, 2020-05-08:
Ok...I'm just providing troubleshooting information for devs.

lurch, 2020-10-01:
Apparently this issue is now being tracked in #3293

dekiesel, 2025-01-28:
So there is no solution but to use root?

tredondo, 2025-06-07:
@dekiesel `sudo balena-etcher` didn't work either (blank screen) with the current version (2.1.12) - blank screen.

Still a problem on Ubuntu 24.04, and adapting the solution from https://askubuntu.com/questions/1512287/obsidian-appimage-the-suid-sandbox-helper-binary-was-found-but-is-not-configu/ didn't work because the balena-etcher .zip isn't an appimage file.