balena-io/etcher #3133: cli resolve symlinks /opt/balenaEtcher/balena-etcher-electron

orcus-de, 2020-05-01:
- **Etcher version: 1.5.76 deb-install** 
- **Operating system and architecture: linux mint 19.3 amd64** 
- **Image flashed:---** 
- **Do you see any meaningful error information in the DevTools?---** 

the deb-file install is missing to resolve symlinks at
`/opt/balenaEtcher/balena-etcher-electron`

which leads to error-messages if using the cli:
```
demo@mint-19-3-cinnamon-64bit:~$ balena-etcher-electron 
/usr/bin/balena-etcher-electron: line 3: /usr/bin/balena-etcher-electron.bin: No such file or directory

```
you should resolve symlinks at the shell script wrapper - like this:
```
#!/bin/bash
#bugfix resolve symlinks, orcus 2020-05-01
o_p=$(readlink $BASH_SOURCE)
if [[ $EUID -ne 0 ]] || [[ $ELECTRON_RUN_AS_NODE ]]; then
  "${o_p%/*}"/balena-etcher-electron.bin "$@"
else
  "${o_p%/*}"/balena-etcher-electron.bin "$@" --no-sandbox
fi
```

lurch, 2020-05-01:
Was that fixed by #3077 ? (I see you're not using the latest version of Etcher)
Or is this a new similar-but-different problem?

orcus-de, 2020-05-01:
I did the use the current version (installed today 2020-05-01) offered by the ppa at linux mint 19.3 64bit.
Does look like the same issue  (didn't look at closed ones).

lurch, 2020-05-01:
I just had a look, and the version at https://bintray.com/etcher/debian is 1.5.80, so I guess you need to follow the instructions at https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 to use the official repo, rather then "the ppa at linux mint 19.3 64bit" ? (and / or you could ask Mint to update their ppa?)

orcus-de, 2020-05-01:
I did exactly use the the shown instructions ... there isn't a "specific ppa for mint 19.3 64bit"

lurch, 2020-05-01:
An alternative might be to download the deb file directly from https://github.com/balena-io/etcher/releases and install it manually? :man_shrugging:

lurch, 2020-05-01:
@zvin @thundron I just did a bit of poking, and I've also "only" got `balena-etcher-electron` version `1.5.76` available on my Ubuntu 16.04 PC.

I'm not very familiar with debian packaging, but I _think_ the line [here](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) of `echo "deb https://deb.etcher.io stable etcher"` means that `apt update` downloads https://dl.bintray.com/etcher/debian/dists/stable/:Release ? That in turn points to `etcher/binary-amd64/Packages` and when I look at https://dl.bintray.com/etcher/debian/dists/stable/etcher/binary-amd64/:Packages it also agrees that the latest-available version is `1.5.76` (which matches what `apt` tells me).
That's as far as my guess-work takes me, so I dunno why that doesn't match up with the `1.5.80` that https://bintray.com/etcher/debian/etcher claims should be available? :confused: 

Is this still a follow-on from the issues that were being discussed in #3078 ?

thundron, 2020-05-04:
It shouldn't... We actually solved that issue long ago (though we're still waiting for an answer to our latest communications).
Closing that one and keeping this one open to investigate

zvin, 2020-05-04:
@orcus-de this was fixed in 1.5.79: https://github.com/balena-io/etcher/pull/3077/commits/d8cb8f78154910f46b70f4b2537d57169b1a0b60
You can get the latest releases from https://github.com/balena-io/etcher/releases until we fix the repository.

orcus-de, 2020-05-04:
cool - thx for the prompt response -  added it to my notes / will share it 
(I should probably have mentioned that I am not an real user of Balena Etcher, but noticed the issue while doing support at the #linuxmint-help  irc help-channel => just sharing the observation and the underlying reason)

cheers orcus

closed 2020-05-06 by zvin

lurch, 2020-05-27:
The bintray links are now failing again - should #3078 be opened again? @thundron

thundron, 2020-05-29:
@lurch Uhm, are they failing with the same error as reported? We didn't change anything so far since the last time this happened

lurch, 2020-05-29:
@thundron Ooops, my mistake, sorry. Previously the bintray URLs I was navigating automatically had a colon in them (as pasted into my message from 1st May above), and when I clicked on those links 2 days ago I got a "The requested path was not found." message (so I assumed they were failing again) ; but if I take out the colons the URLs then work as expected! :tada: (looks like bintray must have tweaked the way directory-indexes get served to web-browsers)

_However_,  https://dl.bintray.com/etcher/debian/dists/stable/etcher/binary-amd64/Packages still lists the latest version as 1.5.76, whilst https://bintray.com/etcher/debian/etcher lists the latest version as 1.5.80 and https://www.balena.io/etcher/ lists the latest version as 1.5.94 :dizzy_face:

lurch, 2020-07-01:
@thundron I've just followed the instructions [here](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) again (on Ubuntu 18.04.4 LTS), and I _still_ only get balena-etcher-electron 1.5.76 installed from the repo, even though I see [bintray](https://bintray.com/etcher/debian/etcher) now thinks 1.5.99 should be available? :confused:

thundron, 2020-07-01:
@lurch I have no idea why that happens :/ have you tried clearing all previous data first? E.g. removing the repo, keys, package, etc.

lurch, 2020-07-01:
@thundron This is the first time I've _ever_ tried installing Etcher onto this particular laptop, so there isn't any "previous data" to remove :confused: 

My `/etc/apt/sources.list.d/balena-etcher.list` says:
```
deb https://deb.etcher.io stable etcher
```

The output of `sudo apt update` includes:
```
Ign:32 https://dl.bintray.com/etcher/debian stable InRelease                   
Get:38 https://dl.bintray.com/etcher/debian stable Release [3,674 B] 
```

`apt policy balena-etcher-electron` says:
```
balena-etcher-electron:
  Installed: 1.5.76
  Candidate: 1.5.76
  Version table:
 *** 1.5.76 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
        100 /var/lib/dpkg/status
     1.5.51 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.5.19 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.4.9 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
```

and this _almost_ matches up with:
```
curl -s https://dl.bintray.com/etcher/debian/dists/stable/etcher/binary-amd64/Packages | grep Version | sort -r
Version: 1.5.76
Version: 1.5.51
Version: 1.5.19
Version: 1.4.9
Version: 1.4.4
Version: 1.4.3
Version: 1.3.1
Version: 1.2.1
Version: 1.2.0
Version: 1.1.2
Version: 1.1.1
Version: 1.0.0~rc.5-1
Version: 1.0.0~rc.4-1
Version: 1.0.0~rc.3-1
Version: 1.0.0~rc.2-1
Version: 1.0.0~rc.1-1
Version: 1.0.0~beta.19-1
Version: 1.0.0~beta.18-1
Version: 1.0.0~beta.17-1
Version: 1.0.0~beta.16-1
```

thundron, 2020-07-02:
@lurch can you try again now? I think I found the issue, it was a naming convention gone wrong

lurch, 2020-07-02:
Fantastic! I've now got balena-etcher-electron 1.5.100 installed from the bintray repo! :confetti_ball: 

```
$ apt policy balena-etcher-electron 
balena-etcher-electron:
  Installed: 1.5.100
  Candidate: 1.5.100
  Version table:
 *** 1.5.100 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
        100 /var/lib/dpkg/status
     1.5.76 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.5.51 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.5.19 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.4.9 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
```