balena-io/etcher #3138: flashing a .wic.bz2 creates an unbootable SD card, but .wic works

PatrickLang, 2020-05-04:
- **Etcher version:** 
1.5.81

- **Operating system and architecture:** 
Windows 10 version 1909 (18363.778)

- **Image flashed:** 
Yocto created image for Raspberry Pi
- **Do you see any meaningful error information in the DevTools?** 
No

WIC support (used by Yocto, OpenEmbedded projects) was added in #2560. By default, Yocto 3.x will create .wic.bz2 files for Raspberry Pi instead of -sdimg. 

If I try to write `core-image-base-raspberrypi0-wifi-20200503052032.rootfs.wic.bz2` with Etcher, it's unbootable. If I expand the .bz2 then write core-image-base-raspberrypi0-wifi-20200503052032.rootfs.wic - it works.

zvin, 2020-05-04:
@PatrickLang Can you please provide a link to your bz2 file ?

PatrickLang, 2020-05-07:
@zvin sure - https://drive.google.com/open?id=1KBTQ6wJzZqzjJuyr2gmKnRFHstHJDMvD

zvin, 2020-05-12:
@PatrickLang this should be fixed in `v1.5.87`, thanks for reporting this.

closed 2020-05-12 by zvin