balena-io/etcher #3139: Flash drive works when prepared with Rufus, but it doesn't with Etcher

brunces, 2020-05-04:
Guys, I have an old computer and, every time I want to install a Linux OS (any distro) on it, I have to prepare a bootable flash drive using **Rufus** for one single reason, it has a section called **Show advanced drive properties** and, under this section, there's an option called **Add fixes for old BIOSes (extra partition, align, etc.)**. I always mark this option and it works every time with any Linux distro I choose to install on that computer.

One thing that sometimes works for me as well is the Linux Mint USB Image Writer, but it does not work with all Linux distro. I have no idea why.

With Etcher, it's the opposite, it simply won't work with any Linux distro I choose. I mean, I am able to prepare the bootable flash drive with Etcher, normally. The problem is, it just won't boot on that old computer.

Just to be clear, Etcher works for me, flawlessly, with any other computers, newer ones. It just won't work, specifically, with that old computer I mentioned above, then I have to use Rufus which is a real pain because I don't use Windows so, every time I need to format that old computer, I need to use a VM or prepare the bootable flash drive at work, where there are Windows machines.

Does Etcher have any option like that, hidden somewhere, that I'm missing?

Thanks for yor time, guys.

brunces, 2020-05-04:
I forgot to mention... If I prepare the flash drive using Rufus, without marking that "Add fixes for old BIOSes (extra partition, align, etc.)" option, it just won't work either. That's why I'm positive that option makes a difference. I'm not an expert, so I can't explain why, but that's just the way it is. Thanks.

lurch, 2020-05-04:
AFAIK Etcher just writes the image you give it directly to the target drive, it doesn't have any options for modifying the image before or after writing it.

thundron, 2020-05-04:
As mentioned above (thank you @lurch!), Etcher doesn't manipulate the image in any way (by choice), so I fear the only option for that old laptop is to keep using Rufus + its legacy option. I wouldn't know of any other solution.

closed 2020-05-04 by thundron