balena-io/etcher #3141: AppImage fails on Debian Buster AMD64

pioruns, 2020-05-05:
- **Etcher version:**
1.5.81-x64
- **Operating system and architecture:** 
Debian GNU/Linux 10 (buster) 5.4.0-0.bpo.4-amd64

- **Image flashed:** 
Didn't get to that point

- **Do you see any meaningful error information in the DevTools?** 
Didn't get to that point

Fails to run AppImage on fully updated, Debian 10 Buster.
Error when running from terminal:

```
$ ./balenaEtcher-1.5.81-x64.AppImage 
[8403:0505/152759.218892:FATAL:setuid_sandbox_host.cc(157)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenaZpHcx0/chrome-sandbox is owned by root and has mode 4755.
/tmp/.mount_balenaZpHcx0/balena-etcher-electron: line 10:  8403 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"
```

lurch, 2020-05-05:
Duplicate of #3129 ?

Hierosme, 2020-05-08:
Yes.

That is the trouble #3129 Etcher consider require root acces for mount somthing in /tmp.

The user never recive a message , when he double click  Etcher icon.

The user have to install sudo, configure sudo, open a terminal , and finally start Etcher.

In that case, use **dd** is more simple, it for that i have repport.

closed 2020-09-28 by thundron