balena-io/etcher #3146: Binaries are out of date on the etcher.io website!

mindfulsource, 2020-05-11:
Please fix the download links on https://www.balena.io/etcher/ to reflect the latest stable ones as per the repo.

lurch, 2020-05-11:
Strange, I'm seeing v1.5.86 at both https://www.balena.io/etcher/ and https://github.com/balena-io/etcher/releases ?

mindfulsource, 2020-05-11:
Weird, it's showing the older version 1.5.45 in my regular Chrome, (even after shift-cmd-r to do a force refresh), but in Safari shows the latest version. Could it be a Google DNS issue??

thundron, 2020-05-11:
We had an issue but that's been resolved and you should see the latest version. You could try cleaning the website data and reloading, that usually helps (or as you mentioned, try with another browser as it may have different configurations - fwiw I can see latest version with Chrome).
Other than that there's nothing we can do on the website itself

closed 2020-05-11 by thundron

mindfulsource, 2020-05-26:
I really hope this doesn't bother you, but at my end the assets are _still_ 1.5.45 links in my regular Chrome browser on my regular macOS machine! I only report this out of concern for others. Are you sure the issue is resolved? Perhaps contact Google / look into DNS more? I'm in Australia. Perhaps Google's DNS in certain regions is having the problem?

I have not done anything weird with my DNS in either my Chrome browser, my system, or my home network. I set Google's DNS at my macOS level (8.8.8.8/8.8.4.4) and at my router level it's just my ISP's default DNS incoming by default. Who knows what Chrome does by default, it probably hijacks and uses Google's DNS inside the browser anyway.

It's not normal in 2020 for web users to have to do website data cleaning as a default thing. AFAIK browsers look after this type of stuff...I haven't wittingly done anything with Chrome extensions that have anything to do with the `balena.io` website or domain.

lurch, 2020-05-26:
Do you see any errors in the javascript console on your chrome browser when viewing the Etcher website?

mindfulsource, 2020-05-26:
Thanks for the tip! Looking at console made me see it may be an extension. Testing in Incognito, the symptom went away, so I found the suspect: It's tampermonkey, and specifically, a userscript called "Anti-Adblock Killer | Reek".

Its github [states](https://github.com/reek/anti-adblock-killer/wiki) that project has been abandoned, so, I've disabled that user script and problem solved. Thanks!

**Lesson learned:** Extensions can definitely ruin your websites by default, so always check those as a vector.