balena-io/etcher #3147: Option to manually exclude some drives from Select Drive

kennu, 2020-05-11:
- **Etcher version:** 1.5.86
- **Operating system and architecture:** Windows 64-bit

I use Etcher a lot (thanks for making it!) to burn custom images on Windows. My problem is that my external SSD drive and my external Storages Spaces RAID devices appear in the target selection and I would like to hide them. I only want to burn images on SD cards and I don't like having to always select the right SD device among the SSD and RAID drives. I'm afraid sometime I'll make a mistake.

My feature request is to somehow manually exclude those unwanted drives from the Etcher target menu. This could happen by editing a configuration file, or by creating a ".etcherignore" hidden file on the drives, or whatever makes best sense.

Here's a screenshot of the target menu I get every time I burn something. You might notice that the "large drive" warning is also unhelpful since I'm using a 64GB SDXC card.

[image]

lurch, 2020-05-11:
I've no idea if it's still included in Etcher, or how you'd set it, but I remember the Etcher developers talking about a "blacklist" feature at one point? https://github.com/balena-io/etcher/search?q=BLACKLIST&type=Commits :man_shrugging: :shrug: 

See #2638 and #1728 and #2071 for similar discussions...

And see also #2116

kennu, 2020-05-12:
Thanks @lurch I figured it out. I added this to my `%APPDATA%\balena-etcher\config.json`:

    {
      ...
      "driveBlacklist": [
        "\\\\.\\PhysicalDrive2",
        "\\\\.\\PhysicalDrive5"
      ]
    }

I guess this issue can be closed since the feature already exists.

Btw, if anybody is wondering, you can list your drives with `wmic diskdrive list brief`. But the name is case-sensitive in Etcher and must be in CamelCase like in my example.

closed 2020-05-12 by kennu