balena-io/etcher #3150: how do i restore my USB stick to its original state

raza-hacks, 2020-05-14:
Because when I go to restore my USB Stick it shows me 3 GB and 700kb (yes the 3 GB and 70kb are showing up as 2 separate drives but connect and disconnect together) and the USB stick is 64 GB
I downloaded Linux on the USB stick

lurch, 2020-05-14:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives :question:

rocciuto, 2020-05-15:
same problem here, used balena to flash a usb memory, the usb become unserviceable. tried the method above but not solved

septriputraw, 2020-06-13:
try format using android phone, i have solved my drive with usb otg & android 10 (aosp) on Asus MPM1

thundron, 2020-06-30:
Please refer to the link by @lurch for solutions on the 3 main OSes.

closed 2020-06-30 by thundron