balena-io/etcher #3154: Checksum failed.

NicTanghe, 2020-05-15:
pretty sure it worked tho.
[image]

rocciuto, 2020-05-15:
same problem here, now usb is useless

lcsvcn, 2020-06-08:
try this:
open cmd and run these commands
1) diskpart
2) list disk
3) select disk <number_of_your_disk>
4) clean
5) create partition primary
6) assign

The usb will appear again and will ask for format.

I am facing the same issue, the usb is not useless, but I cannot flash correctly :/

lcsvcn, 2020-06-08:
I think this might be a bug, since I see a lots of report of people who tried several times in a row with same hardware, device and image and one time run successfully.

rocciuto, 2020-06-08:
thanks for reply, unfortunately the metod you posted doesn't work in my case. 
when I type clean, enter
I get
Diskpart error: Access denied.

usb still dead

NicTanghe, 2020-06-09:
Also, didn't work for me had to download a weird sketchy 3d party disk partitioner, That one worked.

lurch, 2020-06-12:
Related to #3197 ?

thundron, 2020-06-12:
@lurch yep

closed 2021-10-19 by thundron