balena-io/etcher #3157: Packaging Etcher as a snap app

gantonayde, 2020-05-19:
Hi guys, I have packed Etcher as a snap. The snap cannot run with the normal `strict` confinement because it needs to use `pkexec` or `kdesudo` when you click on 'Flash!'. It seems there is no available plug to overcome this but the app works fine with `classic` confinement.

I am interested in uploading and maintaining the snap port (if snapcrafters approve the classic confinement). If you are happy with that, do you have any preferences for the name of the snap:
`etcher`
`balena-etcher`
`balena-etcher-electron`?

lurch, 2020-05-19:
See also #415 where snaps were previously discussed.

gantonayde, 2020-05-19:
Yep, I have seen the thread and @thundron reply. That's why I decided to ask if they are still happy for other people to publish Etcher as a snap and if they want a specific name for the snap

lurch, 2020-05-19:
> Yep, I have seen the thread and @thundron reply.

Cool :+1:  I mentioned the other issue because IMHO it's nice to keep all related discussions inter-linked for future reference :slightly_smiling_face:

thundron, 2020-05-19:
@gantonayde for consistency with the deb/rpm packages I'd say `balena-etcher-electron` would be better :)

lurch, 2020-05-19:
IIRC the name `etcher-electron` was originally chosen to differentiate it from `etcher-cli`. At some point later `etcher-cli` was then discontinued. When Resin rebranded to Balena, Etcher got rebranded to BalenaEtcher, and the package got renamed to `balena-etcher-electron`.

Here ends the history lesson of why there's 3 parts in the package-name :rofl:

thundron, 2020-05-19:
@lurch thanks for the background :P it does feel a little odd to have "balena-etcher-electron" instead of just "balena-etcher" but we do have plans to bring back the CLI soon™ so having *some* distinction is still accurate (to a certain extent :D)

gantonayde, 2020-05-20:
`balena-etcher-electron` it is then. Snapcraft does not allow you at the moment to delete registered names. So, if you guys, decide at some point that you want to take over the snap release, we will need to ask the snapcraft people to transfer the name to you. That's not a problem though.

gantonayde, 2020-05-20:
I have made a request for classic confinement on the snapcraft forums, hopefully they will agree and grant it access to `pkexec` and `kdesudo`. 

You can find snap file [here](https://github.com/gantonayde/balena-etcher-electron/releases).
If you are interested in testing it, the installation should be pretty straightforward:
`sudo snap install balena-etcher-electron_1.5.89_amd64.snap --dangerous --classic`

--dangerous means it is sideloaded, not from the store and --classic in this case is for access to the `pkexec` and `kdesudo`. Please let me know if you find any bugs. There is a chance it uses Mozilla Firefox instead of your default browser for links but that looks like a snap [bug](https://forum.snapcraft.io/t/default-browser-not-used-from-within-snap/4347) for electron apps using classic confinement.

gantonayde, 2020-05-20:
Bad news, Etcher didn't pass the review process for classic. Here is the reason

> Direct access to sudo/pkexec is currently an unsupported use case for classic. Updating the application to use an external helper like udisks2 is the best way to go.

thundron, 2020-05-29:
It's unfortunate :/
Maybe in the future there will be an easier way, but for now we have to rely on those packages

closed 2020-05-29 by thundron

lurch, 2020-05-29:
@thundron or @zvin Please feel free to delete this comment if you don't want a "competitor program" being mentioned in your GitHub issues.

People wanting a Snap may be interested in https://snapcraft.io/rpi-imager :wink:

thundron, 2020-05-31:
@lurch nah it's all good, it's toxic behaviour and constant harassment that are not welcome ;) our docs point to other applications as well [in case of other needs](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#how-can-i-configure-persistent-storage)

xlmnxp, 2020-09-16:
> Bad news, Etcher didn't pass the review process for classic. Here is the reason
> 
> > Direct access to sudo/pkexec is currently an unsupported use case for classic. Updating the application to use an external helper like udisks2 is the best way to go.

not need to be `classic`
I think etcher can work fine in `strict` base `confinement`