balena-io/etcher #3160: Etcher opens to blank screen Win 10 Pro

rklinstiver, 2020-05-20:
- **Etcher 1.5.90
- **Windows 10 Pro 64 bit (Version 2004 Build 19041.264)
I downloaded the latest version of Etcher on my Win10 Pro machine and it opens to a blank screen. Opening as Admin does nothing and running portable version also does nothing.
[Error.log](https://github.com/balena-io/etcher/files/4658739/Error.log)

Ironman273, 2020-05-20:
I have the same issue
* Etcher 1.5.90
* Windows 10 Pro 64 bit (Version 1909 Build 18363.836)

unlinkedmedia, 2020-05-20:
same issue but running 

- Mac OS 10.15.5 Beta (19F83c)

Lewih, 2020-05-20:
Same issue

- Mac OS 10.14.6

dronenb, 2020-05-20:
Same issue
- macOS 10.15.4

Irrelon, 2020-05-20:
Same issue, it flickers to a GUI briefly and then is a blank grey screen. I'm running MacOS Catalina 10.15.2.

Debugging the electron app I can see a stack trace: 

```
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:448 Uncaught TypeError: Cannot read property 'path' of undefined
    at isSourceDrive (drive-constraints.ts:53)
    at Object.getDriveImageCompatibilityStatuses (drive-constraints.ts:145)
    at DriveCompatibilityWarning (target-selector.tsx:32)
    at renderWithHooks (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:14803)
    at mountIndeterminateComponent (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:17482)
    at beginWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:18596)
    at HTMLUnknownElement.callCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:188)
    at HTMLUnknownElement.wrapped (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:444)
    at Object.invokeGuardedCallbackDev (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:292)
    at beginWork$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:23203)
    at performUnitOfWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22154)
    at workLoopSync (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22130)
    at performSyncWorkOnRoot (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21756)
    at /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11089
    at unstable_runWithPriority (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/scheduler/cjs/scheduler.development.js:653)
    at runWithPriority$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11039)
    at flushSyncCallbackQueueImpl (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11084)
    at flushSyncCallbackQueue (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11072)
    at scheduleUpdateOnFiber (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21199)
    at Object.enqueueSetState (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:12639)
    at MainPage.Component.setState (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react/cjs/react.development.js:471)
    at MainPage.tsx:92
    at changeHandler (store.ts:442)
    at Object.dispatch (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/redux/lib/redux.js:228)
    at Object.setDrives (available-drives.ts:26)
    at setDrives (app.ts:196)
    at Scanner.addDrive (app.ts:208)
```

I'm debugging in electron at the moment to see why this is happening...

Irrelon, 2020-05-20:
So one of the components in the react render tree is calling a function with an undefined image. It seems to me that these functions should present a default blank object if they are ok with being called without a defined image object.

Irrelon, 2020-05-20:
I've found the fix, at line 62 of lib/shared/drive-constraints.ts, insert code:

`image = image || {};`

Irrelon, 2020-05-20:
Hey folks, never coded in TypeScript before and never contributed to this project before... just fixing some software I needed to use to flash a linux iso :) - hopefully I haven't stepped on any toes by creating that pull request. Cheers!

AkdM, 2020-05-20:
Same issue here with v1.5.90 using macOS 10.15.4.
I need to insert the SD card only when I click on "Select target". Otherwise, a blank screen appears.

Irrelon, 2020-05-20:
My PR fixed the issue and points to the underlying problem.

Nikostito, 2020-05-20:
Same issue Win 10 Pro. If I open task manager there are 6 instances created before etcher stop responding. Falling back to the previous version fixes it. Also the portable version doesn't even come up as a blank window.

AkdM, 2020-05-20:
> My PR fixes the issue and points to the underlying problem. 

Yep but that was more as a workaround 👍

amitbashan, 2020-05-20:
Same issue here, Windows 10 Home Edition. GUI worked at first however after choosing an ISO it suddenly disappeared. Reverted back to Portable-1.5.89 and it's working.

Irrelon, 2020-05-20:
> > My PR fixes the issue and points to the underlying problem.
> 
> Yep but that was more as a workaround 👍

It’s not really a workaround. In the code the devs have made a point of commenting that passing a blank image object is ok when calling functions where no image is currently present.

Unless I misunderstand you and you mean something you mentioned is a workaround?

Irrelon, 2020-05-20:
> > My PR fixes the issue and points to the underlying problem.
> 
> Yep but that was more as a workaround 👍

Ahh I see :)

outapzaza, 2020-05-21:
Same issuse version 1.5.90 blank screen win 10 pro

thundron, 2020-05-21:
@Irrelon don't worry, I'd say the total opposite :) you're one of the few that added his contribution (around 99% of the commits are from Balena employees) but we'd like many more to come! so thank you very much

The issue is that the image shouldn't be undefined *at all* during that part, so we'll need to check what goes wrong before that, but the PR is totally fine to temporarily fix the issue and get it all up again

PR #3165 (2020-05-21): Fix undefined image from DriveCompatibilityWarning
Fixes: #3160
Change-type: patch
Changelog-entry: Fix undefined image from DriveCompatibilityWarning
commit 31bd8ce7 (2020-05-21): Fix undefined image from DriveCompatibilityWarning

Fixes: #3160
Change-type: patch
Changelog-entry: Fix undefined image from DriveCompatibilityWarning
closed 2020-05-21 by thundron