balena-io/etcher #3161: Flash options not showing in GUI

HeartlandAgTech, 2020-05-20:
- **Etcher version:** 1.5.9
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** Maybe.  I copied 3 errors below

C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:448 Uncaught TypeError: Cannot read property 'path' of undefined
    at isSourceDrive (gui.js:1730)
    at Object.getDriveImageCompatibilityStatuses (gui.js:1822)
    at DriveCompatibilityWarning (gui.js:5060)
    at renderWithHooks (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:14803)
    at mountIndeterminateComponent (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:17482)
    at beginWork (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:18596)
    at HTMLUnknownElement.callCallback (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:188)
    at HTMLUnknownElement.wrapped (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:444)
    at Object.invokeGuardedCallbackDev (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:237)
    at invokeGuardedCallback (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:292)
    at beginWork$1 (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:23203)
    at performUnitOfWork (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:22154)
    at workLoopSync (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:22130)
    at performSyncWorkOnRoot (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:21756)
    at C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11089
    at unstable_runWithPriority (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\scheduler\cjs\scheduler.development.js:653)
    at runWithPriority$1 (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11039)
    at flushSyncCallbackQueueImpl (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11084)
    at flushSyncCallbackQueue (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11072)
    at scheduleUpdateOnFiber (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:21199)
    at Object.enqueueSetState (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:12639)
    at MainPage.Component.setState (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react\cjs\react.development.js:471)
    at gui.js:3898
    at changeHandler (gui.js:785)
    at Object.dispatch (C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\redux\lib\redux.js:228)
    at Object.setDrives (gui.js:2182)
    at setDrives (gui.js:3439)
    at Scanner.addDrive (gui.js:3451)
C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 The above error occurred in the <DriveCompatibilityWarning> component:
    in DriveCompatibilityWarning (created by TargetSelector)
    in p (created by styled.div)
    in styled.div
    in Unknown (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by WithTheme(Component))
    in WithTheme(Component) (created by Styled(WithTheme(Component)))
    in Styled(WithTheme(Component)) (created by TargetSelector)
    in TargetSelector
    in div
    in div
    in Unknown (created by MainPage)
    in div (created by MainPage)
    in div (created by styled.div)
    in styled.div (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by Box)
    in Box (created by Flex)
    in Flex (created by MainPage)
    in div (created by StyledGrommet)
    in StyledGrommet (created by Grommet)
    in Grommet (created by Styled(Grommet))
    in Styled(Grommet) (created by Provider)
    in BreakpointProvider (created by Provider)
    in Provider
    in Unknown (created by MainPage)
    in MainPage

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
console.<computed> @ C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58
C:\Users\RussellSanders\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 {"stack":"TypeError: Cannot read property 'path' of undefined\n    at isSourceDrive (file:///C:/Users/RussellSanders/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:1730:19)\n    at Object.getDriveImageCompatibilityStatuses (file:///C:/Users/RussellSanders/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:1822:9)\n    at DriveCompatibilityWarning (file:///C:/Users/RussellSanders/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:5060:55)\n    at renderWithHooks (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:14803:18)\n    at mountIndeterminateComponent (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:17482:13)\n    at beginWork (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:18596:16)\n    at HTMLUnknownElement.callCallback (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:188:14)\n    at HTMLUnknownElement.wrapped (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\raven-js\\dist\\raven.js:444:21)\n    at Object.invokeGuardedCallbackDev (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:237:16)\n    at invokeGuardedCallback (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:292:31)\n    at beginWork$1 (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:23203:7)\n    at performUnitOfWork (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:22154:12)\n    at workLoopSync (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:22130:22)\n    at performSyncWorkOnRoot (C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:21756:9)\n    at C:\\Users\\RussellSanders\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:11089:24\n    at unstable_runWithPriority (C:\\Users\\RussellSanders\\AppDa [... cut ...] ules\\redux\\lib\\redux.js:228:7)\n    at Object.setDrives (file:///C:/Users/RussellSanders/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:2182:19)\n    at setDrives (file:///C:/Users/RussellSanders/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:3439:21)\n    at Scanner.addDrive (file:///C:/Users/RussellSanders/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:3451:5)","message":"Cannot read property 'path' of undefined"}

rpitting, 2020-05-20:
Same here on mac OS 10.15.4:

`Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.90, Type = dmg
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/resin-corvus/src/resin-corvus.js:39 Wed May 20 2020 21:05:47 GMT+0200 (Mitteleuropäische Sommerzeit) Application start ({"packageType":"dmg","version":"1.5.90","sample":0.1,"applicationSessionUuid":"324a13fe-6275-4e59-ae11-ca3f80f7355c","flashingWorkflowUuid":"f1cde7ed-b98b-42f5-8597-a1f6b93bcc84"})
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:448 Uncaught TypeError: Cannot read property 'path' of undefined
    at isSourceDrive (drive-constraints.ts:53)
    at Object.getDriveImageCompatibilityStatuses (drive-constraints.ts:145)
    at DriveCompatibilityWarning (target-selector.tsx:32)
    at renderWithHooks (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:14803)
    at mountIndeterminateComponent (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:17482)
    at beginWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:18596)
    at HTMLUnknownElement.callCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:188)
    at HTMLUnknownElement.wrapped (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:444)
    at Object.invokeGuardedCallbackDev (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:292)
    at beginWork$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:23203)
    at performUnitOfWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22154)
    at workLoopSync (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22130)
    at performSyncWorkOnRoot (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21756)
    at /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11089
    at unstable_runWithPriority (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/scheduler/cjs/scheduler.development.js:653)
    at runWithPriority$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11039)
    at flushSyncCallbackQueueImpl (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11084)
    at flushSyncCallbackQueue (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11072)
    at scheduleUpdateOnFiber (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21199)
    at Object.enqueueSetState (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:12639)
    at MainPage.Component.setState (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react/cjs/react.development.js:471)
    at MainPage.tsx:92
    at changeHandler (store.ts:442)
    at Object.dispatch (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/redux/lib/redux.js:228)
    at Object.setDrives (available-drives.ts:26)
    at setDrives (app.ts:196)
    at Scanner.addDrive (app.ts:208)
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 The above error occurred in the <DriveCompatibilityWarning> component:
    in DriveCompatibilityWarning (created by TargetSelector)
    in p (created by styled.div)
    in styled.div
    in Unknown (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by WithTheme(Component))
    in WithTheme(Component) (created by Styled(WithTheme(Component)))
    in Styled(WithTheme(Component)) (created by TargetSelector)
    in TargetSelector
    in div
    in div
    in Unknown (created by MainPage)
    in div (created by MainPage)
    in div (created by styled.div)
    in styled.div (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by Box)
    in Box (created by Flex)
    in Flex (created by MainPage)
    in div (created by StyledGrommet)
    in StyledGrommet (created by Grommet)
    in Grommet (created by Styled(Grommet))
    in Styled(Grommet) (created by Provider)
    in BreakpointProvider (created by Provider)
    in Provider
    in Unknown (created by MainPage)
    in MainPage

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
console.<computed> @ /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58
[... cut ...] n    at Object.setDrives (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:2182:19)\n    at setDrives (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:3439:21)\n    at Scanner.addDrive (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:3451:5)","message":"Cannot read property 'path' of undefined"}
console.<computed> @ /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58`

HeartlandAgTech, 2020-05-20:
I removed the SD Card from my PC, and was able to get the GUI to show the click through menu, but as soon as I inserted and SD card, the click through menu would disappear.  I quit and relaunched the tool with out an SD card installed, selected an image, then selected the target to open the selection menu, inserted the drive, and I as able to select the SD card and initiate a flash.

deviprsd, 2020-05-20:
It is being cause by this code https://github.com/balena-io/etcher/blob/ac51e6aae33e773bd1dd75ec5ece42984df76f1c/lib/shared/drive-constraints.ts#L63
The `image` variable needs to be checked for undefined

thundron, 2020-05-20:
We're on it, thanks for the reports. Looks like something goes wrong when getting image infos, resulting in that error

thundron, 2020-05-21:
Should be fixed by PR #3165

lurch, 2020-05-21:
@thundron Fixed now?

thundron, 2020-05-21:
@lurch Yes, but the CI has some issues releasing the new package which we're working on now

zvin, 2020-05-21:
fixed in 1.5.91

closed 2020-05-21 by zvin