balena-io/etcher #3164: Flashing Android OS, Disk Full Error (Even though clearly not)

anthonyb47, 2020-05-21:
The flash is interrupted during decompression, Etcher says the disk image is full but its clearly not (128 GB free). While the process is running i can see the hard drive storage filling up (via refreshing the files explorer), and when the hard drive fills up is when I get the error message. I'm new to this and it's really frustrating lol. Thanks

- **Etcher version:** Lastest
- **Operating system and architecture:**  Dell Intel Laptop
- **Image flashed:** Android Lineage OS
- **Do you see any meaningful error information in the DevTools?**  NO

shawaj, 2020-12-13:
@anthonyb47 do you have more details about what version of Etcher your are using?

And what device are you trying to flash the image to?

lurch, 2020-12-14:
> and when the hard drive fills up is when I get the error message.

When you're flashing a compressed image, Etcher first has to decompress the image (to a temporary location on your hard drive) before you can flash that decompressed image to your destination drive. So it sounds like you don't have enough free space on your hard drive for the image to be fully decompressed? So maybe the "Disk Full Error" is referring to your hard drive, rather than the destination drive you're flashing the image to? I guess a temporary workaround might be to manually decompress the image yourself, and then use Etcher to flash the decompressed image rather than the compressed image?

@shawaj If my hypothesis is correct, maybe the fix here is just to make that particular error message clearer about _which_ disk is full?
(I'm not aware of all the details, but IIRC Etcher used to do "on the fly" decompression-while-flashing which would have avoided this particular problem; but in #3287 @thundron told me that Etcher has been changed to fully decompress the image before flashing)