balena-io/etcher #3167: MacOS not showing GUI content

zhchaoxing, 2020-05-21:
- **Etcher version:** 
1.5.90
- **Operating system and architecture:** 
MacOS Catalina 10.15.4
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** 

After installing Etcher, it shows a blank GUI. It shows the same GUI after restart.

[image]

Not sure if any of the following errors are related
`
Version = 1.5.90, Type = dmg
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/resin-corvus/src/resin-corvus.js:39 Thu May 21 2020 21:27:08 GMT+0800 (China Standard Time) Application start ({"packageType":"dmg","version":"1.5.90","sample":0.1,"applicationSessionUuid":"d68c8727-7f33-4065-8771-c6db92da3e21","flashingWorkflowUuid":"feae2408-f983-421b-9e3d-576d0bbacd03"})
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:448 Uncaught TypeError: Cannot read property 'path' of undefined
    at isSourceDrive (drive-constraints.ts:53)
    at Object.getDriveImageCompatibilityStatuses (drive-constraints.ts:145)
    at DriveCompatibilityWarning (target-selector.tsx:32)
    at renderWithHooks (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:14803)
    at mountIndeterminateComponent (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:17482)
    at beginWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:18596)
    at HTMLUnknownElement.callCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:188)
    at HTMLUnknownElement.wrapped (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:444)
    at Object.invokeGuardedCallbackDev (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:292)
    at beginWork$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:23203)
    at performUnitOfWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22154)
    at workLoopSync (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22130)
    at performSyncWorkOnRoot (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21756)
    at /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11089
    at unstable_runWithPriority (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/scheduler/cjs/scheduler.development.js:653)
    at runWithPriority$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11039)
    at flushSyncCallbackQueueImpl (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11084)
    at flushSyncCallbackQueue (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11072)
    at scheduleUpdateOnFiber (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21199)
    at Object.enqueueSetState (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:12639)
    at MainPage.Component.setState (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react/cjs/react.development.js:471)
    at MainPage.tsx:92
    at changeHandler (store.ts:442)
    at Object.dispatch (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/redux/lib/redux.js:228)
    at Object.setDrives (available-drives.ts:26)
    at setDrives (app.ts:196)
    at Scanner.addDrive (app.ts:208)
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 The above error occurred in the <DriveCompatibilityWarning> component:
    in DriveCompatibilityWarning (created by TargetSelector)
    in p (created by styled.div)
    in styled.div
    in Unknown (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by WithTheme(Component))
    in WithTheme(Component) (created by Styled(WithTheme(Component)))
    in Styled(WithTheme(Component)) (created by TargetSelector)
    in TargetSelector
    in div
    in div
    in Unknown (created by MainPage)
    in div (created by MainPage)
    in div (created by styled.div)
    in styled.div (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by Box)
    in Box (created by Flex)
    in Flex (created by MainPage)
    in div (created by StyledGrommet)
    in StyledGrommet (created by Grommet)
    in Grommet (created by Styled(Grommet))
    in Styled(Grommet) (created by Provider)
    in BreakpointProvider (created by Provider)
    in Provider
    in Unknown (created by MainPage)
    in MainPage

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
console.<computed> @ /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58
/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:58 {"stack":"TypeError: Cannot read property 'path' of undefined\n    at isSourceDrive (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:1730:19)\n    at Object.getDriveImageCompatibilityStatuses (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:1822:9)\n    at DriveCompatibilityWarning (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:5060:55)\n    at renderWithHooks (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:14803:18)\n    at mountIndeterminateComponent (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:17482:13)\n    at beginWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:18596:16)\n    at HTMLUnknownElement.callCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:188:14)\n    at HTMLUnknownElement.wrapped (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/raven-js/dist/raven.js:444:21)\n    at Object.invokeGuardedCallbackDev (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:237:16)\n    at invokeGuardedCallback (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:292:31)\n    at beginWork$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:23203:7)\n    at performUnitOfWork (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22154:12)\n    at workLoopSync (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:22130:22)\n    at performSyncWorkOnRoot (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:21756:9)\n    at /Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11089:24\n    at unstable_runWithPriority (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/scheduler/cjs/scheduler.development.js:653:12)\n    at runWithPriority$1 (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11039:10)\n    at flushSyncCallbackQueueImpl (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11084:7)\n    at flushSyncCallbackQueue (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/react-dom/cjs/react-dom.development.js:11072:3)\n    at scheduleUpdateOnFiber (/Applications/balenaEtcher.app/Conte [... cut ...] t Object.dispatch (/Applications/balenaEtcher.app/Contents/Resources/app/node_modules/redux/lib/redux.js:228:7)\n    at Object.setDrives (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:2182:19)\n    at setDrives (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:3439:21)\n    at Scanner.addDrive (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:3451:5)","message":"Cannot read property 'path' of undefined"}

`

lurch, 2020-05-21:
Fixed in #3165 I believe?

zvin, 2020-05-21:
fixed in 1.5.91

closed 2020-05-21 by zvin