balena-io/etcher #3168: Windows GUI is Blank, can't use Etcher

zhchaoxing, 2020-05-21:
- **Etcher version:** 1.5.9
- **Operating system and architecture:** Windows 10 pro 1909 x64  
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** 

Same issue as https://github.com/balena-io/etcher/issues/3167 but for Windows.

I tried Etcher on Mac, not usable. Tied Etcher on Windows, same issue. Can't use Etcher now. Where can I find the older releases of Etcher?

[image]

Uncaught TypeError: Cannot read property 'path' of undefined
    at isSourceDrive (gui.js:1730)
    at Object.getDriveImageCompatibilityStatuses (gui.js:1822)
    at DriveCompatibilityWarning (gui.js:5060)
    at renderWithHooks (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:14803)
    at mountIndeterminateComponent (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:17482)
    at beginWork (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:18596)
    at HTMLUnknownElement.callCallback (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:188)
    at HTMLUnknownElement.wrapped (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:444)
    at Object.invokeGuardedCallbackDev (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:237)
    at invokeGuardedCallback (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:292)
    at beginWork$1 (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:23203)
    at performUnitOfWork (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:22154)
    at workLoopSync (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:22130)
    at performSyncWorkOnRoot (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:21756)
    at C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11089
    at unstable_runWithPriority (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\scheduler\cjs\scheduler.development.js:653)
    at runWithPriority$1 (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11039)
    at flushSyncCallbackQueueImpl (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11084)
    at flushSyncCallbackQueue (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:11072)
    at scheduleUpdateOnFiber (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:21199)
    at Object.enqueueSetState (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react-dom\cjs\react-dom.development.js:12639)
    at MainPage.Component.setState (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\react\cjs\react.development.js:471)
    at gui.js:3898
    at changeHandler (gui.js:785)
    at Object.dispatch (C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\redux\lib\redux.js:228)
    at Object.setDrives (gui.js:2182)
    at setDrives (gui.js:3439)
    at Scanner.addDrive (gui.js:3451)
C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 The above error occurred in the <DriveCompatibilityWarning> component:
    in DriveCompatibilityWarning (created by TargetSelector)
    in p (created by styled.div)
    in styled.div
    in Unknown (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by WithTheme(Component))
    in WithTheme(Component) (created by Styled(WithTheme(Component)))
    in Styled(WithTheme(Component)) (created by TargetSelector)
    in TargetSelector
    in div
    in div
    in Unknown (created by MainPage)
    in div (created by MainPage)
    in div (created by styled.div)
    in styled.div (created by ForwardRef)
    in ForwardRef (created by Component)
    in Component (created by ForwardRef)
    in ForwardRef (created by ForwardRef)
    in ForwardRef (created by Box)
    in Box (created by Flex)
    in Flex (created by MainPage)
    in div (created by StyledGrommet)
    in StyledGrommet (created by Grommet)
    in Grommet (created by Styled(Grommet))
    in Styled(Grommet) (created by Provider)
    in BreakpointProvider (created by Provider)
    in Provider
    in Unknown (created by MainPage)
    in MainPage

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
console.<computed> @ C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58
C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 {"stack":"TypeError: Cannot read property 'path' of undefined\n    at isSourceDrive (file:///C:/Users/zhcha/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:1730:19)\n    at Object.getDriveImageCompatibilityStatuses (file:///C:/Users/zhcha/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:1822:9)\n    at DriveCompatibilityWarning (file:///C:/Users/zhcha/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:5060:55)\n    at renderWithHooks (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:14803:18)\n    at mountIndeterminateComponent (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:17482:13)\n    at beginWork (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:18596:16)\n    at HTMLUnknownElement.callCallback (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:188:14)\n    at HTMLUnknownElement.wrapped (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\raven-js\\dist\\raven.js:444:21)\n    at Object.invokeGuardedCallbackDev (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:237:16)\n    at invokeGuardedCallback (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:292:31)\n    at beginWork$1 (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:23203:7)\n    at performUnitOfWork (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:22154:12)\n    at workLoopSync (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:22130:22)\n    at performSyncWorkOnRoot (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:21756:9)\n    at C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\react-dom.development.js:11089:24\n    at unstable_runWithPriority (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\scheduler\\cjs\\scheduler.development.js:653:12)\n    at runWithPriority$1 (C:\\Users\\zhcha\\AppData\\Local\\Programs\\balena-etcher\\resources\\app.asar\\node_modules\\react-dom\\cjs\\ [... cut ...] s/balena-etcher/resources/app.asar/generated/gui.js:2182:19)\n    at setDrives (file:///C:/Users/zhcha/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:3439:21)\n    at Scanner.addDrive (file:///C:/Users/zhcha/AppData/Local/Programs/balena-etcher/resources/app.asar/generated/gui.js:3451:5)","message":"Cannot read property 'path' of undefined"}
console.<computed> @ C:\Users\zhcha\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58

lurch, 2020-05-21:
> Where can I find the older releases of Etcher?

https://github.com/balena-io/etcher/releases

mrkag, 2020-05-21:
I can get around this by not mounting the SD card until after Etcher is started. 
Unmount SD card -> Start Etcher -> Select File -> Mount SD Card (Etcher recognizes automatically) -> Select Flash

zvin, 2020-05-21:
fixed in 1.5.91

closed 2020-05-21 by zvin