balena-io/etcher #3171: v1.5.91 refuses to load multiple times on ubuntu.

Ntemis, 2020-05-22:
- **Etcher version:** 
v1.5.91
- **Operating system and architecture:** 
X86_64 Ubuntu 18.04.4
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 
Doesnt run

No

When run from command line i get this
------------------------------------------------------

(balena-etcher-electron.bin:20023): Gtk-WARNING **: 23:55:39.672: Theme parsing error: gtk.css:6765:28: Missing opening bracket in color definition
(node:20023) UnhandledPromiseRejectionWarning: Error: ENOENT, node_modules/@ronomon/direct-io/binding.node not found in /opt/balenaEtcher/resources/app.asar
    at createError (electron/js2c/asar.js:111:17)
    at Object.func [as .node] (electron/js2c/asar.js:146:27)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
    at require (internal/modules/cjs/helpers.js:68:18)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/etcher-sdk/build/aligned-lockable-buffer.js:3:21)
    at Module._compile (internal/modules/cjs/loader.js:880:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:892:10)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
(node:20023) UnhandledPromiseRejectionWarning: Error: ENOENT, node_modules/@ronomon/direct-io/binding.node not found in /opt/balenaEtcher/resources/app.asar
    at createError (electron/js2c/asar.js:111:17)
    at Object.func [as .node] (electron/js2c/asar.js:146:27)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
    at require (internal/modules/cjs/helpers.js:68:18)
    at Object.<anonymous> (/opt/balenaEtcher/resources/app.asar/node_modules/etcher-sdk/build/aligned-lockable-buffer.js:3:21)
    at Module._compile (internal/modules/cjs/loader.js:880:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:892:10)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
(node:20023) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:20023) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:20023) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:20023) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Ntemis, 2020-05-22:
After apt purged and apt installed v1.5.90 all is fine.
Bug exist only in .91 version

Ntemis, 2020-05-23:
on v1.5.90 the app launches but is an empty gray box, you cant select anything.
Bug is random

lurch, 2020-05-23:
I believe the v1.5.90 release was known-broken (on all OSes?) and got rapidly fixed in #3165 so not a good version to compare to :wink:

Ntemis, 2020-05-27:
new version just released , all ok
thank you

closed 2020-05-27 by Ntemis