balena-io/etcher #3173: UNKNOWN:unknown error read

Aronlele, 2020-05-23:
- **Etcher version:** 1.5.91
- **Operating system and architecture:** windows10 64bit
- **Image flashed:** dmg
- **Do you see any meaningful error information in the DevTools?** 
[image]
[image]
[-1590246557846.log](https://github.com/balena-io/etcher/files/4672011/-1590246557846.log)

lurch, 2020-05-23:
Sounds similar to #3110 ?

Aronlele, 2020-05-24:
> Sounds similar to #3110 ?

i don`t know ，hope have a help

Aronlele, 2020-05-29:
anyone else can help me？

zvin, 2020-05-29:
@Aronlele this means we couldn't read from the flashed drive for verification.
In general it mean the drive isn't working properly. Try flashing another drive.

closed 2020-05-29 by zvin

Aronlele, 2020-05-29:
> @Aronlele this means we couldn't read from the flashed drive for verification.
> In general it mean the drive isn't working properly. Try flashing another drive.
I have changed several devices, the one I used at first was able to write normally, but now all devices are not allowed

lurch, 2020-05-29:
I wonder if "it used to work but now it doesn't" means that a Windows security setting has been changed somewhere? I don't know the full details, but it seems that sometimes "blocks" part of Etcher? :shrug:  (and apologies if this comment is totally irrelevant!)

Aronlele, 2020-06-01:
> @Aronlele what do you mean with "are not allowed"?

all devices failed

Aronlele, 2020-06-01:
> I wonder if "it used to work but now it doesn't" means that a Windows security setting has been changed somewhere? I don't know the full details, but it seems that sometimes "blocks" part of Etcher? 🤷 (and apologies if this comment is totally irrelevant!)

Thank u ，but I I run Etcher with administrator rights ， @~@

thundron, 2020-06-02:
@Aronlele can you try checking Windows's "Controlled folder access" setting?

thundron, 2020-06-02:
Although it's unlikely because it *should* pop-up as an `EPERM` error iirc, it's worth checking since it worked before

OutLore, 2022-03-22:
I know this thread is a couple of years old, but I just had the same issue. It was solved by re-installing Etcher with a rightclick-runas admin command. (User was logged on as an admin account, but not the Administrator)... 
When the OP says all devices failed I suspect they mean that when Etcher fails, it leaves the USB devices in an unusable state, not formatted or with a partition, so Windows cannot see them, they have to be manually re-partitioned.