balena-io/etcher #3175: [Suggestion][Discussion] Aren't releases too frequent?

molaeiali, 2020-05-25:
Hey there, I am maintaining [etcher-bin](https://aur.archlinux.org/packages/etcher-bin/) package on Arch Linux AUR
Don't you think the project releases are so frequent? Is really 2 or 3 commits worth a release tag? Most of the times there is no noticeable change but people are feeling behind and want to update quickly! It also is making keeping the package up-to-date harder, sometimes there are 2 or 3 release tags in one day! [Image](http://imgur.com/gallery/zpwhQZ9) 

Don't you want to think of some milestones to reach before a release? or at least some key feature changes or bug fixes before a release?

What's the rush to have releases this frequent?

lurch, 2020-05-25:
I believe that Balena are using the https://en.wikipedia.org/wiki/Continuous_deployment model.

yepyepyep4711, 2020-05-25:
Continuous Deployment totally leaves room for a stable and a beta branch. Also, automated testing is a part of that process, and that clearly is not happening in enough environments, if at all (I have no way of knowing, I'm just looking at the results). Look at all the people complaining about the GUI not loading, for example. It seems the dev team can't reproduce the problem, which means they don't have enough testing resources. Which is not an accusation, really, just the way it is. But I would suggest they indeed adapt their release cycle to that fact and slow down. I've seen this software advertised as "the best" everywhere. And last month I was using it too. Now on one machine it won't install at all (stays stuck at roughly one third of the progress bar), and on the other the GUI won't load. So I'm back with Win32DiskImager (both machines are win7 64bit). Looks less cool, but works.

lurch, 2020-05-25:
A long time ago there _were_ separate beta and stable release channels for Etcher (see the mention of **snapshot** and **production** release types in https://github.com/balena-io/etcher/blob/master/docs/PUBLISHING.md - which is now outdated) but it seems that feature got dropped at some point.
(looks like that was part of https://github.com/balena-io/etcher/commit/3855bb4d56ac57a2309f47da74217f55d049d281#diff-b67911656ef5d18c4ae36cb6741b7965 )

> Sometimes there is a release, and 1 hour later a bug fix for last release, why?

Yeah, it seems that https://github.com/balena-io/etcher/blob/master/docs/MAINTAINERS.md#testing is unfortunately being skipped sometimes. But Etcher only has a small team of developers and AIUI they work on other Balena projects too.

molaeiali, 2020-05-25:
Speaking of two releases in one day, today 1.5.92 and right now 1.5.93 😅 I am now really expecting 1.5.94; under 24 hours 🚶‍♂️🚶‍♂️🚶‍♂️

thundron, 2020-05-26:
It's something that we haven't discussed yet again, but we changed for a reason. We implemented the merge=release process with an auto-updater so that we'd have control over the releases - let them sit in the wild a bit before dispatching the update to the full userbase - but we might discuss this again as we're also not 100% happy with how the auto-updater works

molaeiali, 2020-08-19:
I don't know if it was any changes in project's side or not, but I see the releases are now less frequent, so I close the issue

closed 2020-08-19 by molaeiali