balena-io/etcher #3178: Unable to run etcher

GerBreOwn, 2020-05-26:
- **Etcher version:** 1.5.93-x64
- **Operating system and architecture:** Linux Mint Debian Edition (LMDE)
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Downloaded and extracted balena-etcher-electron-1.5.93-linux-x64.zip to balenaEtcher-1.5.93-x64.Appimage. Not able to run that program. I also  did the echo to the "/apt/sources.list" and then updated "apt" and did "sudo apt-get install balena-etcher-electron". Still can NOT get etcher to run. I have previously used ETCHER for many years without any problems. Looks like this new version has problems.
Thanks.

thundron, 2020-05-29:
@GerBreOwn If you run it from the console, what logs do you see? Is there any error?

displaced1, 2020-05-31:
balenaEtcher-1.5.94-x64.AppImage: 19: balenaEtcher-1.5.94-x64.AppImage: Syntax error: Unterminated quoted string
is the error I get when attempting to run etcher from the command line

thundron, 2020-05-31:
@displaced1 what's the command you run? It may be that you forgot to close a string running from command line (i.e. you forgot an ending quote)

displaced1, 2020-05-31:
Download, make sure it's set to execute then click on it in file explorer. Open at the command line with sh ./file name and I get the error you see.

lurch, 2020-06-01:
Etcher isn't a script, you shouldn't be running it with `sh` :wink: