balena-io/etcher #3179: Installation problem in newer versions

r1versEdge, 2020-05-27:
- **Last couple releases, ever since beginning of year** 
- **Kali Linux, amd64** 
- **No image, problem with installation** 
- **Required libpango1.0-0 to be installed, libpango is installed, it's asking for libpango1.0-0 when it should be asking for libpango-1.0-0 in the dependencies, this makes it so it won't install because of broken dependencies, one of the developers must have misspelled the dependency so now it won't install or work**

lurch, 2020-05-27:
Duplicate of #3119 which has been fixed. What version of Etcher are you trying to use?

Looks like there's still problems with the bintray repo though (see the discussion in #3133 ), so try downloading the latest `.deb` from https://github.com/balena-io/etcher/releases

r1versEdge, 2020-05-27:
Ok, thanks for the help

closed 2020-05-29 by thundron