balena-io/etcher #3181: why only run as root on Linux

btwise, 2020-05-28:
Through continuous attempts, it can be built normally under Linux, and dep, RPM and appimage binaries are also automatically packaged, but it can only be run with root permissions, and the chrome-sandbox will not automatically set 4755 permissions, so the ordinary account cannot be run. What needs to be modified?

lurch, 2020-05-28:
Duplicate of #3129 ?

https://github.com/balena-io/etcher/issues/3119#issuecomment-622989049 might have a fix?

btwise, 2020-05-28:
If there is no solution, it should be solved from the project script, rather than the user constantly typing the command

zvin, 2020-05-29:
Etcher works fine as a regular user, it'll only elevate the writer process when needed.
What do you mean when you say "it can only be run with root permissions" ?

btwise, 2020-05-29:
> Etcher works fine as a regular user, it'll only elevate the writer process when needed.
> What do you mean when you say "it can only be run with root permissions" ?
I mean is: if the etcher does not work properly without sudo, it will tell you that the chrome-shadbox does not have 755 permissions, and if it runs sudo, it will work fine

lurch, 2020-05-29:
@zvin The other issues I linked to suggest this might only be an issue on Debian 10 ? :shrug:

thundron, 2020-05-29:
What distro of Linux are you trying to run Etcher on?
What version of Etcher?

closed 2020-09-28 by thundron

cow-firmware, 2022-01-05:
I was running this on centOS, only could get the balena etcher window to appear when running as root, content was empty.