balena-io/etcher #3185: BalenaEtcher makes USB Sticks unusable (RAW format)

Pflegusch, 2020-05-29:
- **Etcher version: 1.5.76** 
- **Operating system and architecture: Windows 10 64-Bit** 
- **Image flashed: https://github.com/foxlet/bootra1n - Version 0.10.2** 
- **Do you see any meaningful error information in the DevTools?
[image]
[image]** 

So I tried flashing the image above using BalenaEtcher version 1.5.76. I used 3(!) different USB Sticks from which I know everyone was working fine because I use them on the regular, but after I try to boot from the usb sticks nothing happens and if I boot back into Windows the USB sticks seem to be in an unusable state. Windows Disk Manager states that they are in RAW format and I can't format them anymore or reflash them whatsoever (see errors above). Even Testdisk is unable to to fix this problem so if anyone can please help me this would be really appreciated. Spend too much hours now and I can't help myself any more.

lurch, 2020-05-29:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives :question:

Pflegusch, 2020-05-29:
> 
> 
> https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives ❓

Trying that right now, didn't see that. Thanks :)

closed 2020-05-29 by Pflegusch