balena-io/etcher #3194: win10 x86 not showing GUI content

aptx17, 2020-06-01:
- **Etcher version:** v1.5.95/1.5.94/1.5.93/1.5.92/1.5.91/1.5.90
- **Operating system and architecture:** Windows 10 LTSB 10240
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** 

This issue with 32bit winOS.
[image]

> electron/js2c/asar.js:140 Uncaught Error: \\?\C:\Users\zxc\Desktop\app-32\resources\app\generated\modules\@ronomon\direct-io\binding.node is not a valid Win32 application.
\\?\C:\Users\zxc\Desktop\app-32\resources\app\generated\modules\@ronomon\direct-io\binding.node
    at process.func (electron/js2c/asar.js:140)
    at process.func [as dlopen] (electron/js2c/asar.js:140)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034)
    at Object.func (electron/js2c/asar.js:140)
    at Object.func [as .node] (electron/js2c/asar.js:140)
    at Module.load (internal/modules/cjs/loader.js:815)
    at Module._load (internal/modules/cjs/loader.js:727)
    at Module._load (electron/js2c/asar.js:738)
    at Function.Module._load (electron/js2c/asar.js:738)
    at Module.require (internal/modules/cjs/loader.js:852)
    at require (internal/modules/cjs/helpers.js:74)
    at Object.<anonymous> (gui.js:43062)
    at __webpack_require__ (gui.js:16)
    at Object.<anonymous> (gui.js:35291)
    at __webpack_require__ (gui.js:16)
    at Object.<anonymous> (gui.js:52820)

zvin, 2020-06-02:
Thanks for reporting this @aptx17 , I'll have a a look.

zvin, 2020-06-03:
This should be fixed in 1.5.96 @aptx17

zvin, 2020-06-03:
nevermind, 1.5.96 is broken on windows

zvin, 2020-06-09:
this should be fixed in 1.5.97

closed 2020-06-09 by zvin