balena-io/etcher #3197: Lock the drive to prevent failure on verification

sajolida, 2020-06-03:
Hi Etcher, sajolida from [Tails](https://tails.boum.org/) behind the keyboard.

We've been recommending Etcher to install Tails since 2019 and we're now serving close to 1000 downloads of Etcher every day! We chose Etcher because it is very easy to use and cross-platform.

So first of all, thanks a bunch!  :love_you_gesture:

People have been reporting verification failures on our GitLab: https://gitlab.tails.boum.org/tails/tails/-/issues/17634

And I've tracked them down to this thread on your forum: https://forums.balena.io/t/checksums-do-not-match/36537

To summarizes this thread, on some Windows installations a partition on the device is being opened in the background and Windows creates a "System Volume Information" folder in it. As a result, the checksum verification fails.

As per [comment 38](https://forums.balena.io/t/checksums-do-not-match/36537/38) by @zvin, I understand that the solution to these failures would be to lock the drive on Windows.

I couldn't find any issue related to this on your GitHub, so here is one :) I hope it helps and I hope it's feasible.

- **Etcher version:** I tried both 1.5.79 and 1.5.81. People on your forum reported this issue as early as 1.5.57.
- **Operating system and architecture:** Windows 10 64-bit.
- **Image flashed:** Tails, both ISO and USB images in my case. Raspberry Buster on the thread on your forum.
- **Do you see any meaningful error information in the DevTools?** I can't reproduce this issue myself.

thundron, 2020-06-05:
@sajolida thanks for the report :) I wouldn't know how to lock access to a drive on Windows, as far as I know there's a setting on Windows for that which doesn't look like it's easy to circumvent (more on that [here](https://superuser.com/questions/1199823/how-to-prevent-creation-of-system-volume-information-folder-in-windows-10-for/1199824#1199824)) so this might be something up to the user to do. We can add it to the docs probably since it affects quite many users

lurch, 2020-06-05:
https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-fsctl_lock_volume ?